ompl/ompl #1145: libompl.so cannot be found

frederiknolte, 2024-02-22:
I installed using the wheel on Ubuntu 20.04, Python 3.9. It is not possible to import ompl as the file `libompl.so` cannot be found. After manually searching for it, I found a file `libompl.so.17` in `~/.anaconda3/envs/<env_name>/lib/python3.9/site-packages`. By manually specifying the location of the `libompl.so.17` file in the `dll_loader` function in `~/.anaconda3/envs/<env_name>/lib/python3.9/site-packages/ompl/__init__.py`, I got to circumvent the error:

Instead of:
```
sys = system()
if sys == 'Windows':
    ext = '.dll'
elif sys == 'Darwin':
    ext = '.dylib'
else: # Linux, other UNIX systems
    ext = '.so'
fname = path + '/lib' + lib + ext
if not isfile(fname):
    fname = find_library(lib)
```

I have:
```
sys = system()
if sys == 'Windows':
    ext = '.dll'
elif sys == 'Darwin':
    ext = '.dylib'
else: # Linux, other UNIX systems
    ext = '.so'
fname = path + '/lib' + lib + ext
if not isfile(fname):
    fname = find_library(lib)
if not isfile(fname):  # Dirty fix
    fname = '/' + join(*path.split('/')[:-2]) + '/lib' + lib + ext + '.17'
```

kylc, 2024-02-28:
It seems like you're mixing libraries between the wheel and a separate version you've installed from Anaconda.

Can you try reproducing the issue in a minimal environment, without clutter from other OMPL installation sources? For example, When I install the OMPL wheel in a Python 3.11 virtualenv called "venv", I see the shared libraries at this path:

```
$ ls venv/lib/python3.11/site-packages/libompl.so*
venv/lib/python3.11/site-packages/libompl.so
venv/lib/python3.11/site-packages/libompl.so.1.6.0
venv/lib/python3.11/site-packages/libompl.so.17
```

Thodoris1999, 2026-04-09:
Our wheels have been redesigned (available from `pip install ompl`) and should behave correctly now as they pass our python tests from an isolated environment