ompl/ompl #1149: Setting up start and goal state in CompoundStateSpace()

akash-venkateshwaran, 2024-03-16:
Hello,
I am trying to define CompoundStateSpace and create the start and goal state in this space using python. Here is the code and the error.
Kindly request your help

Code:
def _init_simple_setup(self, ompl_state: OMPLPathState) -> og.SimpleSetup:
        space = ob.CompoundStateSpace()

        se2_space = ob.SE2StateSpace()

        # Create the 1D state space for velocity
        vel_space = ob.RealVectorStateSpace(1)  # 1D for velocity
        # Set bounds for the velocity
        velocity_bounds = ob.RealVectorBounds(1)
        velocity_bounds.setLow(ompl_state.domain_Vel[0])  # Set minimum velocity
        velocity_bounds.setHigh(ompl_state.domain_Vel[1])  # Set maximum velocity, adjust as needed
        vel_space.setBounds(velocity_bounds)

        bounds = ob.RealVectorBounds(2)
        bounds.setLow(0, ompl_state.domain_XY[0])
        bounds.setLow(1, ompl_state.range_XY[0])
        bounds.setHigh(0, ompl_state.domain_XY[1])
        bounds.setHigh(1, ompl_state.range_XY[1])
        se2_space.setBounds(bounds)

        space.addSubspace(se2_space, 1.0)  
        space.addSubspace(vel_space, 1.0)  

        print(f"state space bounds: x=[{bounds.low[0]}, {bounds.high[0]}]; y=[{bounds.low[1]}, {bounds.high[1]}]")

        simple_setup = og.SimpleSetup(space)
        simple_setup.setStateValidityChecker(ob.StateValidityCheckerFn(self.is_state_valid))

        print("\n",space)  # Just to check what the space looks like
        print("\n",type(space))  # Should show something like <class 'ompl.base.CompoundStateSpace'>

        start = space.allocState()
        goal = space.allocState()

        start[0].setX(ompl_state.ship_position_XY[0])
        start[0].setY(ompl_state.ship_position_XY[1])
        start[1][0] = 0  # Set initial velocity in the start state

        goal[0].setX(ompl_state.goal_position_XY[0])
        goal[0].setY(ompl_state.goal_position_XY[1])
        goal[1][0] = 5 

        simple_setup.setStartAndGoalStates(start, goal)

Error:
File "/workspaces/MELO_workspace/build/path_planning/path_planning/ompl_path.py", line 430, in _init_simple_setup
    simple_setup.setStartAndGoalStates(start, goal)
Boost.Python.ArgumentError: Python argument types in
    SimpleSetup.setStartAndGoalStates(SimpleSetup, CompoundStateInternal, CompoundStateInternal)
did not match C++ signature:
    setStartAndGoalStates(ompl::geometric::SimpleSetup {lvalue}, ompl::base::ScopedState<ompl::base::StateSpace> start, ompl::base::ScopedState<ompl::base::StateSpace> goal, double threshold=2.220446049250313e-16)
[ros2run]: Process exited with failure 1

mamoll, 2024-03-16:
Try this:
```python
    start = ob.State(space) # allocate a ScopedState
    goal = ob.State(space)

    # operator() returns State for ScopedState
    start()[0].setX(ompl_state.ship_position_XY[0])
    start()[0].setY(ompl_state.ship_position_XY[1])
    start()[1][0] = 0  # Set initial velocity in the start state

    goal()[0].setX(ompl_state.goal_position_XY[0])
    goal()[0].setY(ompl_state.goal_position_XY[1])
    goal()[1][0] = 5 
```

akash-venkateshwaran, 2024-03-21:
Thank you very much!

closed 2024-04-23 by zkingston