ompl/ompl #1151: Implementation of Constraints on the SolutionPath

akash-venkateshwaran, 2024-03-21:
Hello,

I am new to OMPL. I am trying to incorporate constraints into path planning. For example, check out the figure where the cost function is such that the farther you move away from the origin, the lower the cost. You can see the optimal path. Now, in this problem, I want to add a constraint that the path length must be smaller than a certain value. 

[image]

I checked the ompl::base::Constraint class, but I don't think we can use it here. This class is meant to provide constraints on your C-space for a single state, not for a sequence of states. I was thinking the only way would be to obtain the PlannerData while solving and somehow apply the constraint. Please let me know if there is any other way

kylc, 2024-03-29:
Hi @akash-venkateshwaran,

I don't believe that OMPL's optimal planning framework supports hard constraints on the path. Instead, if your usecase allows it, you may be able to formulate a multi-objective minimization problem using the existing `OptimizationObjective` framework combined with an optimal planner, like RRT*. Careful weighting of the path length and origin distance costs may get you a satisfactory solution. As an extension, you could try to implement your path length objective using a [penalty method](https://ocw.mit.edu/courses/15-084j-nonlinear-programming-spring-2004/13b4c46d590cf3df94475e0ae1a815c3_lec10_penalty_mt.pdf) to approximate a hard constraint, but I don't know how well this will converge within OMPL.

It also sounds like your problem may be better suited to trajectory optimization approaches, especially if your objective and constraints are convex.

zkingston, 2024-03-29:
One possible trick you could use is to take advantage of the [Prolate Hyperspheroid](https://ompl.kavrakilab.org/classompl_1_1ProlateHyperspheroid.html) to bound the sampling domain to states that would be within a path less than your desired maximum

Read https://arxiv.org/pdf/1706.06454.pdf for some more information.

zkingston, 2024-04-23:
Closing due to lack of activity.

closed 2024-04-23 by zkingston