ompl/ompl #1156: Problem on mukti-agent planning with SE2MultiRigidBodyPlanning

yhiroya, 2024-04-28:
I'm currently using SE2MultiRigidBodyPlanning.cpp for multi-agent planning. When I increase the number of agents from the default 2 to 5, even if collisions with obstacles occur, "Exact Solution" is still output, and the path planning is judged as successful. 
 I have created obstacle and agent files in accordance with the default settings for SE2MultiRigidBodyPlanning.cpp.
 I'd like to know the cause of this issue. Also, can this program truly be extended to handle more than 3 agents?

zkingston, 2024-04-28:
Can you describe the output exact solution path? If the robots are "teleporting" through obstacles or each other this is probably an issue with the valid segment resolution. See https://ompl.kavrakilab.org/stateValidation.html for more information (the paragraph on validity checking resolution) for how to tune these values.

yhiroya, 2024-04-30:
I'm sorry. 
Due to my mistake, the program wasn't functioning properly, but after fixing it, I was able to run SE2MultiRigidBodyPlanning.cpp correctly even with three or more instances. 
I apologize for any inconvenience caused.

closed 2024-04-30 by yhiroya