ompl/ompl #1166: Would it be possible to add Python wheels for linux arm64 as well?

BolunDai0216, 2024-06-10:
Hi,

I noticed that the Python wheels for arm64 are only for MacOS, it would also be great to have the wheels for linux arm64 as well!

kylc, 2024-09-23:
To implement this right now, we would have to cross-compile for arm64 from our x86_64 Linux runners.

It is probably simpler to wait until Linux/arm64 runners become available for open source on Github (supposedly EOY2024). Follow: https://github.com/orgs/community/discussions/19197

BolunDai0216, 2024-09-23:
> To implement this right now, we would have to cross-compile for arm64 from our x86_64 Linux runners.
> 
> It is probably simpler to wait until Linux/arm64 runners become available for open source on Github (supposedly EOY2024). Follow: https://github.com/orgs/community/discussions/19197

@kylc would it be possible to provide the commands to compile the python wheels so that we can try it on our own arm64 devices?

kylc, 2024-09-24:
I don't have any arm64 devices, so I can't test.

You can try using the `cibuildwheel` tool as we use in CI. For example, to build for my system with Python 3.11 on x86_64, I use the command `CIBW_BUILD="cp311-manylinux_x86_64" cibuildwheel py-bindings/`.

You may be able to do something like `CIBW_BUILD="cp311-manylinux_aarch64" cibuildwheel py-bindings/` on your arm64 system.