ompl/ompl #1167: Error when building OMPL Python Bindings on aarch64

BolunDai0216, 2024-06-10:
Hi,

I was trying to build the python bindings, and I encounter this error:

```bash
ERROR error occured, while parsing element with name "Field" and attrs "['id', 'name', 'type', 'context', 'access', 'offset']".
Error: 'line'.
Error: can't generate code for module control
```

the entire output is:

```bash
> cmake -DPYTHON_EXEC=/usr/bin/python3 ..
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
-- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: serialization filesystem system program_options 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Found PY_numpy: /usr/lib/python3/dist-packages/numpy  
-- Found PY_pyplusplus: /usr/local/lib/python3.8/dist-packages/pyplusplus  
-- Found PY_pygccxml: /usr/local/lib/python3.8/dist-packages/pygccxml  
-- Found castxml: /usr/bin/castxml  
-- Could NOT find yaml-cpp (missing: yaml-cpp_DIR)
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * boost_numpy (required version == 1.71.0)
 * Python, <https://www.python.org>
   Used for python bindings.
 * boost_python (required version == 1.71.0)
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * boost_unit_test_framework (required version == 1.71.0)

-- The following REQUIRED packages have been found:

 * boost_serialization (required version == 1.71.0)
 * boost_filesystem (required version == 1.71.0)
 * boost_system (required version == 1.71.0)
 * boost_program_options (required version == 1.71.0)
 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * boost_headers (required version == 1.71.0)
[... 431 lines cut ...]
chardet             3.0.4               
dbus-python         1.2.16              
distro-info         0.23+ubuntu1.1      
graphsurgeon        0.4.6               
idna                2.8                 
importlib-metadata  7.1.0               
numpy               1.17.4              
pip                 20.0.2              
pygccxml            2.2.1               
PyGObject           3.36.0              
pyplusplus          1.8.6               
python-apt          2.0.1+ubuntu0.20.4.1
requests            2.22.0              
requests-unixsocket 0.2.0               
setuptools          45.2.0              
six                 1.14.0              
tensorrt            8.5.2.2             
uff                 0.6.9               
unattended-upgrades 0.1                 
urllib3             1.25.8              
wheel               0.34.2              
zipp                3.19.2   
```

Thanks in advance!

iLUVE69, 2024-06-27:
ompl's fucking stupid bro honestly
someone pls get this in pypi already

iLUVE69, 2024-06-27:
@mamoll

patrick-5546, 2024-07-05:
- Related discussion: https://github.com/ompl/ompl/issues/1116#issuecomment-2035894445

johnnynunez, 2025-03-04:
solved:
it is working https://github.com/johnnynunez/ompl/actions/runs/13643981441
close :)

BolunDai0216, 2025-03-04:
@johnnynunez This is awesome! Can you point me to where I can download the wheels?

johnnynunez, 2025-03-04:
> [@johnnynunez](https://github.com/johnnynunez) This is awesome! Can you point me to where I can download the wheels?

Now really

johnnynunez, 2025-03-05:
@BolunDai0216 ompl 1.7.0 is coming, but you can download here:
https://pypi.jetson-ai-lab.dev/jp6/cu128/ompl/1.6.0

saahu27, 2025-03-06:
I am closing this issue, as python bindings are working on the latest commit. We are working on the new release, and will be available soon.

closed 2025-03-18 by saahu27