ompl/ompl #1170: OMPL installation issues on Ubuntu 24.04

iLUVE69, 2024-07-02:
Can anyone help me to install OMPL step by step on Ubuntu 24.04, I've tried everything(running the script from website) checking dependencies and everything. But my system can't seem to generate python bindings

wbthomason, 2024-07-02:
@iLUVE69 If all you need is the Python bindings, you may have an easier time installing a pre-built wheel from the assets here: https://github.com/ompl/ompl/releases/tag/prerelease

wbthomason, 2024-07-02:
Note that you will need to uninstall any system-level installation of OMPL before using the pre-built wheel due to an unfortunate issue with dynamic linkage.

If you need help beyond this, please provide more information: what output do you get from the different stages of the build? What specific versions of dependencies do you have installed? etc.

iLUVE69, 2024-07-02:
hey @wbthomason i'll soon do a new installation after complete cleanup, i will share with you the output log , please help me with what could be the issue, as far as i understand i cant build the python bindings as running a demo throws "cant import util._util " error. I was thinking of downgrading my Ubuntu from 24.04 to 22.04

wbthomason, 2024-07-02:
The issue with building Python bindings is usually a missing dependency, though they are tricky to get working. I can't really suggest other causes for the issue without seeing the output logs. The problem is almost certainly not your Ubuntu version, although we should add 24.04 to the CI.

Again, if you only need to use OMPL through Python (and not e.g. extend it through C++), I highly recommend downloading the appropriate pre-built wheel for your OS and Python version from the link above and just installing it. Usually this is much easier for this use case.

iLUVE69, 2024-07-03:
@wbthomason From what i see, i dont have a update_bindings target in my build/Release directory. How do i make it? i have the dependencies installed, castxml, pygccxml and pypluplus . What am i missing??

wbthomason, 2024-07-03:
Please attach your build logs, in full, so that I can see the problem.

iLUVE69, 2024-07-03:
Here:

ASUSLaptop-X415JA-X415JA:~/ompl/build$ cmake ..
-- Build type: Release
CMake Warning at CMakeModules/FindPython.cmake:197 (message):
  Failed to find matching Boost.Python library! Python bindings probably
  won't work.
Call Stack (most recent call first):
  CMakeLists.txt:94 (find_boost_python)

-- Checking for module 'libspot'
--   Package 'libspot', required by 'virtual:world', not found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
CMake Warning at CMakeModules/FindPython.cmake:220 (message):
  Failed to find matching Boost.Numpy library! Python bindings probably won't
  work.
Call Stack (most recent call first):
  CMakeLists.txt:139 (find_boost_numpy)

CMake Warning at CMakeModules/FindPython.cmake:197 (message):
  Failed to find matching Boost.Python library! Python bindings probably
  won't work.
Call Stack (most recent call first):
  CMakeModules/PythonBindingsUtils.cmake:5 (find_boost_python)
  py-bindings/CMakeLists.txt:1 (include)

-- Found PY_pyplusplus: /home/ismellweed/anaconda3/lib/python3.11/site-packages/pyplusplus  
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * flann (required version >= 1.9.2), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * Python, <https://www.python.org>
   Used for python bindings.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * boost_unit_test_framework (required version == 1.82.0)
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following REQUIRED packages have been found:

 * boost_serialization (required version == 1.82.0)
 * boost_atomic (required version == 1.82.0)
 * boost_filesystem (required version == 1.82.0)
 * boost_system (required version == 1.82.0)
 * boost_program_options (required version == 1.82.0)
 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * boost_headers (required version == 1.82.0)
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following features have been disabled:

 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.

-- The following OPTIONAL packages have not been found:

 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).
 * boost_numpy (required version == 1.82.0)
 * boost_python-py311 (required version == 1.82.0)
 * boost_python (required version == 1.82.0)

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done (0.5s)
-- Generating done (0.1s)
-- Build files have been written to: /home/ismellweed/ompl/build

These are the directory contents:

ASUSLaptop-X415JA-X415JA:~/ompl/build$ ls

bin                      CTestTestfile.cmake      ompl.pc
castxml.cfg              demos                    py-bindings
CMakeCache.txt           doc                      Release
CMakeFiles               install-ompl-ubuntu.sh   scripts
cmake_install.cmake      Makefile                 share
cmake_uninstall.cmake    omplConfig.cmake         src
CPackConfig.cmake        omplConfigVersion.cmake  tests
CPackSourceConfig.cmake  omplExport.cmake

iLUVE69, 2024-07-03:
this is the boost version:

ASUSLaptop-X415JA-X415JA:~/ompl/build$ dpkg -s libboost-dev | grep Version
Version: 1.83.0.1ubuntu2

wbthomason, 2024-07-03:
The failure seems to arise due to these issues:
```shell
Failed to find matching Boost.Numpy library! Python bindings probably won't work.
Call Stack (most recent call first):
CMakeLists.txt:139 (find_boost_numpy)

CMake Warning at CMakeModules/FindPython.cmake:197 (message):
Failed to find matching Boost.Python library! Python bindings probably won't work.
Call Stack (most recent call first):
CMakeModules/PythonBindingsUtils.cmake:5 (find_boost_python) py-bindings/CMakeLists.txt:1 (include)
```

This probably means you're missing `libboost-python-dev` and `libboost-numpy-dev`, though I note that CMake was searching for specifically version 1.82, and you have Boost 1.83 - I'm not sure if that's a problem. Do you have those libraries installed?

iLUVE69, 2024-07-05:
Hey @wbthomason , I finally fixed this by totally removing my Ubuntu 24.04 and installing 22.04. I don't think this was any issue specifically due to Ubuntu 24.04 but there were too many dependency issues to the point it was difficult to fix them. After scraping everything, I decided to run the --app script and everything fell into place. Thanks for the assistance and a happy Independence day:)

closed 2024-07-05 by iLUVE69