ompl/ompl #1173: Is deterministic planning possible?

zkytony, 2024-07-15:
Looking at [RRT.h](https://ompl.kavrakilab.org/geometric_2planners_2rrt_2RRT_8h_source.html), it has an RNG member. But the only place that uses it in the [RRT.cpp source](https://ompl.kavrakilab.org/geometric_2planners_2rrt_2src_2RRT_8cpp_source.html) is goal bias sampling. There are other places that involve randomness, such as the very next line `goal_s->sampleGoal(rstate)`, but after tracing it down I cannot find any use of RNG in those samplers.   For example, In [UniformValidSampler](https://ompl.kavrakilab.org/UniformValidStateSampler_8cpp_source.html), it uses a `sampler_`
  allocated by SpaceInformation via `allocStateSampler`, and the only implementation I found for `allocStateSampler` is in ProjectedStateSampler [here](https://github.com/ompl/ompl/blob/af4d6d625e15a4ed6d85255c718ce26bcd79e4cc/src/ompl/base/spaces/constraint/ProjectedStateSpace.h#L122) but I I don't see RNG or SEED being used by ProjectedStateSampler.

So my conclusion is there is no determinism. Am I missing something?

mamoll, 2024-07-15:
There is a DeterministicSampler class which uses a Halton sequence to generate samples. You have to use StateSpace::setStateSamplerAllocator method to specify a function that returns a configured instance of the DeterministicSampler class. Of course, you can use the same method to return a sampler that does whatever you want. You may also want to look at the PrecomputedStateSampler.

wbthomason, 2024-07-15:
@zkytony to expand a bit on @mamoll's answer: sampling-base motion planning tends to default to using pseudo-random number generators in a few places, but deterministic samplers (such as those based on the Halton sequence) are well-known and also commonly used. OMPL supports both types of samplers out of the box; for a clear example of actually using the default RNG in state sampling, take a look here: https://github.com/ompl/ompl/blob/af4d6d625e15a4ed6d85255c718ce26bcd79e4cc/src/ompl/base/spaces/src/RealVectorStateSpace.cpp#L45-L53

Beyond this, `ompl::RNG` can also be made "deterministic" by setting its seed, e.g. using its single-argument constructor or the `setLocalSeed` method: https://ompl.kavrakilab.org/classompl_1_1RNG.html. This makes the sequence of samples deterministic as `ompl::RNG` uses pseudo-random number generation under the hood.

Does this, combined with @mamoll's response, answer your question?

zkytony, 2024-07-15:
Thanks for the information. I am new to OMPL and not familiar with Halton sequence. But looks like I need to create my own state sampler? My situation may be simpler and I hope to not have to add more complication if possible.   I am using an off-the-shelf models under OMPL APP, following a demo example, such as [this one](https://github.com/ompl/omplapp/blob/main/demos/SE2RigidBodyPlanning/KinematicCarPlanning.cpp). 

Here is a sketch of my code:
```c++
// setup comes from a provided model, e.g. KinematicCarPlanning
setup.setPlanner(std::make_shared<RRT>(setup.getSpaceInformation()));

// in main function
ompl::RNG::setSeed(12345);
setup.solve(10)
```
I hope the fact of adding `ompl::RNG::setSeed(12345)` is enough to make the planner's output deterministic.

wbthomason, 2024-07-15:
I would recommend calling `setSeed` as early as possible in `main`, before any OMPL objects are created - but doing so should suffice to make the output deterministic.

zkytony, 2024-07-15:
Thanks for the confirmation. I observe running the planner several times in a sequence yield different results. But I will report back if I have a concrete example.

closed 2024-07-15 by zkytony

wbthomason, 2024-07-15:
Please do, and feel free to @ me in the issue. I would expect that this is happening because of (1) some statically-constructed object (which may invoke or copy the state of a RNG before you've seeded it), (2) use of a RNG other than `ompl::RNG`, or maybe (3) variations in the order in which you are calling the planners. 

In some cases, if you're seeding the RNG after it's been used, you'll see some error/warning output, e.g. https://github.com/ompl/ompl/blob/main/src/ompl/util/src/RandomNumbers.cpp#L71-L98. Do you? If not, it's likely that some RNG is getting copied before you call `setSeed`, or (2-3).

zkytony, 2024-07-15:
@wbthomason I do get the error message if I call `ompl::RNG::setSeed(12345)` every time before a new planner is created.

>Random number generation already started. Changing seed now will not lead to deterministic sampling.

Therefore I do the call before creating any planner.  

I tried to look into how RNG is used by the RRT planner, but it was not obvious as there are multiple sources of randomness (goal bias, goal sampler, state sampler, etc). It should hopefully be easy enough to verify if (1) and (2) are the case by looking at the source code, right?

mamoll, 2024-07-15:
You may observe non-determinism **even if you fix the random seed** if your code (or the planners you use) are multi-threaded, since thread scheduling happens at the OS-level and can vary from run to run (depending on what else is running on your machine). That's why I recommended DeterministicStateSampler and PrecomputedStateSampler since they don't depend on RNG.

mamoll, 2024-07-16:
As @wbthomason pointed out to me privately, even the DeterministicStateSampler and PrecomputedStateSampler can result in non-deterministic behavior if the planner has multiple threads calling the same sampler.