ompl/ompl #1174: Compile ompl in cmake external project through ExternalProject or FetchContent

icosac, 2024-07-19:
Hi, 

I was trying to see if it was possible to compile and link ompl to an external cmake project I am working on. My idea is to use either FetchContent or ExternalProject from cmake to download and compile ompl and link it against my code. 

The problem is that if I add:

```cmake 
FetchContent_Declare(
  ompl
  GIT_REPOSITORY https://github.com/ompl/ompl.git
  GIT_TAG 1.6.0
)
FetchContent_MakeAvailable(ompl)
```

Then I cannot find the headers when compiling my code. So I added

```cmake
target_include_directories(points PRIVATE ${CMAKE_BINARY_DIR}/_deps/ompl-src/src)
```

Which fixes the missing headers for my files, but it raises the following errors when compiling ompl

```
In file included from /home/user/Projects/PD-line/build/_deps/ompl-src/src/ompl/base/StateSampler.h:41,
                 from /home/user/Projects/PD-line/build/_deps/ompl-src/src/ompl/base/StateSpace.h:42,
                 from /home/user/Projects/PD-line/build/_deps/ompl-src/src/ompl/base/spaces/SE2StateSpace.h:40,
                 from /home/user/Projects/PD-line/build/_deps/ompl-src/src/ompl/base/spaces/DubinsStateSpace.h:40,
                 from /home/user/Projects/PD-line/src/main_read_points.cpp:37:
/home/user/Projects/PD-line/build/_deps/ompl-src/src/ompl/util/RandomNumbers.h:46:10: fatal error: ompl/config.h: No such file or directory
   46 | #include "ompl/config.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/points.dir/build.make:76: CMakeFiles/points.dir/src/main_read_points.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:357: CMakeFiles/points.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
```

I'm not that good with ExternalProject as I'm not sure how to correctly link ompl headers and compiled files to my project. 

Has anyone had any success in linking ompl with FetchContent or ExternalProject? 

I successfully managed to use git modules and `add_subdirectory`, but I would much prefer to have something integrated directly with cmake. Do you reckon it would be possible? 

Thanks!

Ryanf55, 2024-09-08:
It should be possible, we have been overhauling the CMake support of this library. When done well, ompl should be able to be used as
*  build and install in system headers
* build and install in a custom cmake prefix path
* consume through cpack as a debian
* consume through a submodule
* consume through fetchcontent
* consume as a ROS package through colcon

Thank you for supplying your example; I can use that to test against. I'll try to make time to investigate soon.

Ryanf55, 2024-09-09:
I have confirmed your report. Here is a minimal reproducible example. By not specifying the version, it should fetch main.
```cmake
cmake_minimum_required(VERSION 3.22)
project(ompl_consumer)

include(FetchContent)
FetchContent_Declare(
  ompl
  GIT_REPOSITORY https://github.com/ompl/ompl.git
)
FetchContent_MakeAvailable(ompl)

add_executable(main main.cpp)
target_link_libraries(main PRIVATE ompl::ompl)

```

```c++
#include <ompl/base/ScopedState.h>

int main() {
    return 0;
}
```

closed 2025-10-01 by saahu27