ompl/ompl #1175: Can't find `OMPL` when building MoveIt2 and OMPL from source

MarqRazz, 2024-07-24:
I'm working on debugging a motion planning issue with RRTConnect, but I can't seem to get MoveIt & OMPL to build in the same workspace. OMPL builds successfully but MoveIt fails to find the package...

```
--- stderr: moveit_planners_ompl                            
CMake Error at /opt/ros/humble/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'OMPL' was not found
  before
Call Stack (most recent call first):
  ompl_interface/CMakeLists.txt:34 (ament_target_dependencies)
```

You can see in the error it's looking for package `OMPL` but in this repo I see that the package name is `ompl`. You can also see [here](https://github.com/moveit/moveit2/blob/main/moveit_planners/ompl/package.xml#L25) that MoveIt requires the lower case version of package name so I'm not sure where the issue lies.

I've tested this on both Humble and Jazzy and get the same result. I tried this same approach several months ago for a different issue but ran into the same problem so this export/find issue has been around for a while as far as I can tell.

To test this I pulled down the following docker image(s)
`osrf/ros:humble-desktop` or `osrf/ros:jazzy-desktop`
updated the containers to the latest release and then cloned MoveIt (for the selected ROS version) and OMPL from source before building with `colcon build`.

mamoll, 2024-07-24:
Hi @MarqRazz! OMPL is ROS-agnostic and should be treated as a non-ROS cmake-based package (cmake for building OMPL, but also cmake for finding it as a dependency). There is a package.xml in the top-level OMPL that makes it a little easier to be part of the ROS ecosystem, but that's it. By default, it installs its cmake-related files as:
```
${prefix}/share/ompl/cmake/omplConfig.cmake
${prefix}/share/ompl/cmake/omplConfigVersion.cmake
```
so that `find_package(ompl)` should "just work". I am not sure where the uppercase OMPL package name is coming from. The ROS-packaged release of OMPL is managed via this repo: https://github.com/ros-gbp/ompl-release. I don't see it there either.

MarqRazz, 2024-07-24:
The upper case OMPL is coming from [this](https://github.com/moveit/moveit2/blob/main/moveit_planners/ompl/ompl_interface/CMakeLists.txt#L41) MoveIt package. If I change all of the instances of the upper case OMPL to lower then MoveIt fails to find all of the include headers for its interface.

```
--- stderr: moveit_planners_ompl                                               
In file included from /root/moveit2_ws/src/moveit2/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/model_based_planning_context.h:39,
                 from /root/moveit2_ws/src/moveit2/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/planning_context_manager.h:39,
                 from /root/moveit2_ws/src/moveit2/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp:37:
/root/moveit2_ws/src/moveit2/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/parameterization/model_based_state_space.h:39:10: fatal error: ompl-1.6/ompl/base/StateSpace.h: No such file or directory
   39 | #include <ompl/base/StateSpace.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I also can't get MoveIt to build if I purge `ros-$(ROS_DISTRO)-ompl` an then re-install it so the issue is likely in MoveIt and how it pulls in the dependency. I will keep poking around and let you know if I can isolate the difference between binary and source.

mamoll, 2024-07-24:
Can you attach the CMakeCache.txt for moveit_planners_ompl? [It looks like it's trying to find `ompl` as a dependency the right way.](https://github.com/moveit/moveit2/blob/main/moveit_planners/ompl/CMakeLists.txt#L23)

MarqRazz, 2024-07-25:
Thanks for the help @mamoll! Is it common for packages to use find_package in lower case and then link against the library by referencing a different name (i.e. all caps but the same name)? I don't think I have seen this pattern in other libraries before.
```
find_package(ompl REQUIRED)
ament_target_dependencies(some_exe  OMPL Boost ....)
```

Here is the [CMakeCache.txt](https://github.com/user-attachments/files/16369538/CMakeCache.txt).

mamoll, 2024-07-25:
Can you try replacing `OMPL` with `ompl::ompl`? I just remembered that the cmake package code for OMPL changed in the last release to use the modern cmake style.

MarqRazz, 2024-07-25:
I thought the same thing after reading #1067 but get the same error.
```
CMake Error at /opt/ros/jazzy/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:77 (message):
  ament_target_dependencies() the passed package name 'ompl::ompl' was not
  found before
Call Stack (most recent call first):
  ompl_interface/CMakeLists.txt:32 (ament_target_dependencies)
```

MarqRazz, 2024-07-25:
More test results....

Calling `ament_target_dependencies` with lower case `ompl` seems to work but it cant find the include headers. Poking around I found that the cmake variable `${OMPL_INCLUDE_DIRS}` is not set after `find_package(ompl REQUIRED)`. I added the following after [this line](https://github.com/moveit/moveit2/blob/main/moveit_planners/ompl/CMakeLists.txt#L30) in the cmake file of MoveIt and it prints an empty list.
```
message("********************************************")
foreach(dir ${OMPL_INCLUDE_DIRS})
  message("dir='${dir}'")
endforeach()
message("********************************************")
```

If I add that same message to the end of the Cmake file in OMPL it prints
```
dir='/root/moveit2_ws/src/ompl/src'
dir='/root/moveit2_ws/build/ompl/src'
```

MarqRazz, 2024-07-25:
# UPDATE
:confetti_ball: I have successfully built both OMPL and MoveIt2 from source :confetti_ball: 

I had to manually define `OMPL_INCLUDE_DIRS` as shown above and then moved all of the linking from `ament_target_dependencies` to `target_link_libraries`.
Example
```
ament_target_dependencies(
  moveit_ompl_planner_plugin
  moveit_core
  moveit_ros_planning
  rclcpp
  pluginlib
  tf2_ros
  Boost)
target_link_libraries(moveit_ompl_planner_plugin moveit_ompl_interface ompl::ompl)
```

I tried to play around with [these lines](https://github.com/ompl/ompl/blob/main/CMakeLists.txt#L232-L234) in the OMPL CmakeLists.txt file but couldn't figure out the right way to make it so that `ament_target_dependencies` could find the library.

Sorry I'm not the best at Cmake! Do you have any suggestions on how I can get OMPL to properly set the variable `OMPL_INCLUDE_DIRS` and also make it so that colcon/ament can properly find the library?

MarqRazz, 2024-07-25:
Other things I noticed as I have been debugging this.

- Humble has `1.6.0` released but it's ~70 commits behind `main` and does not include PR 1067 which is why it is still working with MoveIt2
- Jazzy only has `1.5.2` released

Once we have a fix for OMPL I would be happy to help get MoveIt updated. Do you have thoughts about how we should go about synchronizing the releasing of both libraries so we don't break things for ROS users?

mamoll, 2024-07-27:
@vrabaud made himself a contributor to https://github.com/ros-gbp/ompl-release recently. Maybe he could create a 1.6.0 release for Jazzy? It'd be even better if an active MoveIt maintainer would take on the job of creating ROS releases of OMPL, since MoveIt is the primary user of OMPL (@henningkayser , @sjahr?).

@MarqRazz's solution seems more like a workaround. Surely, `ament` knows how to deal with modern cmake config files that only define targets, right? How does ament deal with Eigen, another C++ library that uses modern cmake config files?

sjahr, 2024-07-30:
Happy to help, are there any special instructions on how to create a release in OMPL or can I just follow the instructions on the [moveit website](https://moveit.ros.org/documentation/contributing/releases/) and adapt to OMPL? @MarqRazz is a MoveIt maintainer too, couldn't he also get the rights to create an OMPL-ROS release himself, especially given that he is currently working with the OMPL source build directly.

MarqRazz, 2024-07-30:
> @MarqRazz's solution seems more like a workaround. Surely, ament knows how to deal with modern cmake config files that only define targets, right? How does ament deal with Eigen, another C++ library that uses modern cmake config files?

It's a bit of a work around but not by much. According to the [ament documentation](https://docs.ros.org/en/jazzy/How-To-Guides/Ament-CMake-Documentation.html#linking-to-dependencies) 

> Modern CMake prefers to use only targets, exporting and linking against them. CMake targets may be namespaced, similar to C++. Prefer to use the namespaced targets if they are available. For instance, Eigen3 defines the target Eigen3::Eigen.

> In the example of Eigen3, the call should then look like

`target_link_libraries(my_library PUBLIC Eigen3::Eigen)`
> This will also include necessary headers, libraries and their dependencies. Note that this dependency must have been previously discovered via a call to find_package.

Note: `This will also include necessary headers, libraries and their dependencies.`

I have to manually set the path for the headers with ompl so I suspect some export issue because they are in the `build` and `install` locations and I can use either one if I set
```
list(APPEND OMPL_INCLUDE_DIRS "/root/moveit2_ws/src/ompl/src")
list(APPEND OMPL_INCLUDE_DIRS "/root/moveit2_ws/build/ompl/src")
```
or 
```
list(APPEND OMPL_INCLUDE_DIRS "/root/moveit2_ws/install/ompl/include/ompl-1.6")
```
before this line in [moveit_planners/ompl/CMakeLists](https://github.com/moveit/moveit2/blob/main/moveit_planners/ompl/CMakeLists.txt#L30)

MarqRazz, 2024-07-30:
> Surely, ament knows how to deal with modern cmake config files that only define targets, right?

I tried changing [these lines](https://github.com/ompl/ompl/blob/main/CMakeLists.txt#L232-L234) to set the file name to include the name spacing with `${PROJECT_NAME}::${PROJECT_NAME}` but it still doesn't find it.
```
file(WRITE ${CMAKE_CURRENT_BINARY_DIR}/share/ament_index/resource_index/packages/${PROJECT_NAME}::${PROJECT_NAME} "")
install(FILES ${CMAKE_CURRENT_BINARY_DIR}/share/ament_index/resource_index/packages/${PROJECT_NAME}::${PROJECT_NAME} DESTINATION share/ament_index/resource_index/packages)

DuYi-angle, 2024-08-02:
@MarqRazz I've made the same changes according to your modifications, but I still can't compile both moveit2 and ompl source code, maybe I've made a mistake somewhere, because I'm not sure which files you've made those modifications under, can you provide me with a complete modification process, I'll be very grateful!

MarqRazz, 2024-08-02:
To verify what OMPL is setting the environment variable `OMPL_INCLUDE_DIRS` to I added the following to the end of it's CMakeLists.txt file.
```
message("&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&")
foreach(dir ${OMPL_INCLUDE_DIRS})
  message("dir='${dir}'")
endforeach()
message("&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&")
```

On my machine it output
```
dir='/root/moveit2_ws/src/ompl/src'
dir='/root/moveit2_ws/build/ompl/src'
```

I then check what MoveIt thinks the env var `OMPL_INCLUDE_DIRS` by adding the same print function right before it tries to include them [here](https://github.com/moveit/moveit2/blob/main/moveit_planners/ompl/CMakeLists.txt#L30). I get an empty list printed. If I set the variable manually before I print I can see it's set and MoveIt will successfully build against it.
```
list(APPEND OMPL_INCLUDE_DIRS "/root/moveit2_ws/src/ompl/src")
list(APPEND OMPL_INCLUDE_DIRS "/root/moveit2_ws/build/ompl/src")
```

MarqRazz, 2024-08-02:
I then moved all of the `ament_target_dependencies` statements of `OMPL` like [this](https://github.com/moveit/moveit2/blob/main/moveit_planners/ompl/ompl_interface/CMakeLists.txt#L41) to the target_link_libraries line [here](https://github.com/moveit/moveit2/blob/main/moveit_planners/ompl/ompl_interface/CMakeLists.txt#L51). 
`target_link_libraries(moveit_generate_state_database moveit_ompl_interface ompl::ompl)`

Here is a copy of my changes to that file.
[CMakeLists.txt](https://github.com/user-attachments/files/16470704/CMakeLists.txt)

MarqRazz, 2024-08-06:
If I move OMPL out of the colcon workspace, and manually build it with `make`, `make install` I can also get MoveIt to build from source but it also fails to properly link against the library and throws the `Failed to load library` error:
```
[move_group-4]   what():  Failed to load library /root/moveit2_ws/install/moveit_planners_ompl/lib/libmoveit_ompl_planner_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: libompl.so.17: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99
```

Ryanf55, 2024-09-06:
In ROS humble, it's a bit difficult to link correctly. Here's my system info and a working approach to bring it in through binaries. I'm happy to help come up with an approach that should make binaries and source work the same, and be easy.

```
$ echo $ROS_DISTRO
humble
$ rosdep resolve ompl
#apt
ros-humble-ompl
$ apt show ros-humble-ompl
Package: ros-humble-ompl
Version: 1.6.0-1jammy.20240728.192232
Priority: optional
Section: misc
Maintainer: Mark Moll <mmoll@rice.edu>
Installed-Size: 8,807 kB
Depends: libboost-filesystem1.74.0 (>= 1.74.0), libboost-serialization1.74.0 (>= 1.74.0), libc6 (>= 2.34), libgcc-s1 (>= 3.4), libode8 (>= 2:0.16.2), libstdc++6 (>= 11), libboost-all-dev, libeigen3-dev, libflann-dev, libode-dev, ros-humble-ros-workspace
Homepage: https://ompl.kavrakilab.org
Download-Size: 1,755 kB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: OMPL is a free sampling-based motion planning library.
```

To find it:
```
ryan@FRIEDMAN-W10:~/Dev/ompl_ws/tutorials$ dpkg -L ros-humble-ompl | grep ".cmake"
/opt/ros/humble/share/ompl/cmake
/opt/ros/humble/share/ompl/cmake/omplConfig.cmake
/opt/ros/humble/share/ompl/cmake/omplConfigVersion.cmake
```

Here's `omplConfig.cmake`.
```cmake
$ cat /opt/ros/humble/share/ompl/cmake/omplConfig.cmake
# CMake OMPL module
#
# It defines the following variables:
# OMPL_FOUND         - TRUE
# OMPL_INCLUDE_DIRS  - The OMPL include directories
# OMPL_LIBRARY_DIRS  - The OMPL library directories
# OMPL_LIBRARIES     - The OMPL library
# OMPLAPP_LIBRARIES  - The OMPL.app libraries (if installed)
# OMPL_VERSION       - The OMPL version in the form <major>.<minor>.<patchlevel>
# OMPL_MAJOR_VERSION - Major version
# OMPL_MINOR_VERSION - Minor version
# OMPL_PATCH_VERSION - Patch version

####### Expanded from @PACKAGE_INIT@ by configure_package_config_file() #######
####### Any changes to this file will be overwritten by the next CMake run ####
####### The input file was omplConfig.cmake.in                            ########

get_filename_component(PACKAGE_PREFIX_DIR "${CMAKE_CURRENT_LIST_DIR}/../../../" ABSOLUTE)

macro(set_and_check _var _file)
  set(${_var} "${_file}")
  if(NOT EXISTS "${_file}")
    message(FATAL_ERROR "File or directory ${_file} referenced by variable ${_var} does not exist !")
  endif()
endmacro()

####################################################################################

set(OMPL_VERSION 1.6.0)
set(OMPL_MAJOR_VERSION 1)
set(OMPL_MINOR_VERSION 6)
set(OMPL_PATCH_VERSION 0)

set_and_check(OMPL_INCLUDE_DIR "${PACKAGE_PREFIX_DIR}/include/ompl-1.6")
set(OMPL_INCLUDE_DIRS "${OMPL_INCLUDE_DIR};/usr/include;/usr/include/eigen3")
foreach(_dir /usr/include;/usr/include;;TRIANGLE_INCLUDE_DIR-NOTFOUND;;;;)
    if(_dir)
        list(APPEND OMPL_INCLUDE_DIRS "${_dir}")
    endif()
endforeach()
list(REMOVE_DUPLICATES OMPL_INCLUDE_DIRS)
set(OMPL_INCLUDE_DIRS "${OMPL_INCLUDE_DIRS}" CACHE STRING "Include path for OMPL and its dependencies")

set_and_check(OMPL_LIBRARY_DIR ${PACKAGE_PREFIX_DIR}/lib/x86_64-linux-gnu)
set(OMPL_LIBRARY_DIRS "${OMPL_LIBRARY_DIR};/usr/lib/x86_64-linux-gnu")
foreach(_dir /usr/lib/x86_64-linux-gnu;/usr/lib;/usr/lib/x86_64-linux-gnu;;;)
    if(_dir)
        list(APPEND OMPL_LIBRARY_DIRS "${_dir}")
    endif()
endforeach()
list(REMOVE_DUPLICATES OMPL_LIBRARY_DIRS)
set(OMPL_LIBRARY_DIRS "${OMPL_LIBRARY_DIRS}" CACHE STRING "Library path for OMPL and its dependencies")

find_library(OMPL_LIBRARIES NAMES ompl.${OMPL_VERSION} ompl
    PATHS ${OMPL_LIBRARY_DIR} NO_DEFAULT_PATH)
foreach(_lib /usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;ode;)
    if(_lib)
        list(APPEND OMPL_LIBRARIES "${_lib}")
    endif()
endforeach()

find_library(OMPLAPPBASE_LIBRARY NAMES ompl_app_base.${OMPL_VERSION} ompl_app_base
    PATHS ${OMPL_LIBRARY_DIR} NO_DEFAULT_PATH)
find_library(OMPLAPP_LIBRARY NAMES ompl_app.${OMPL_VERSION} ompl_app
    PATHS ${OMPL_LIBRARY_DIR} NO_DEFAULT_PATH)
if (OMPLAPPBASE_LIBRARY AND OMPLAPP_LIBRARY)
    set(OMPLAPP_LIBRARIES "${OMPLAPPBASE_LIBRARY};${OMPLAPP_LIBRARY};;")
    if(OPENGL_LIBRARIES)
        list(APPEND OMPLAPP_LIBRARIES "${_lib}")
    endif()
    set(OMPLAPP_LIBRARIES "${OMPLAPP_LIBRARIES}"
        CACHE STRING "Paths to OMPL.app libraries")
endif()

include(FindPackageHandleStandardArgs)
find_package_handle_standard_args(ompl DEFAULT_MSG OMPL_INCLUDE_DIRS OMPL_LIBRARY_DIRS OMPL_LIBRARIES)
```

It's the old style CMake, there is no `ompl::ompl` target.

To use it, you can do the following after adding this in your `package.xml` as a dependency:
```xml
 <depend>ompl</depend>
 ```
 
 ```cmake
cmake_minimum_required(VERSION 3.22)
project(ompl_simple)

# set(CMAKE_FIND_DEBUG_MODE TRUE)
find_package(ompl CONFIG REQUIRED)

add_executable(main main.cpp)
target_link_libraries(main PRIVATE ${OMPL_LIBRARIES} )
target_include_directories(main PRIVATE ${OMPL_INCLUDE_DIRS} )
```

To make this easier for ROS users, I propose backporting whatever we need to `humble` to expose an `ompl::ompl` target. 
Then, you can just do this instead:
```cmake
find_package(ompl CONFIG REQUIRED)

add_executable(main main.cpp)
target_link_libraries(main PRIVATE ompl::ompl )
```

MarqRazz, 2024-09-06:
@Ryanf55 The binary release of Humble works on my system (just as you described) because the latest release of OMPL does not include [this commit that you made last Oct.](https://github.com/ompl/ompl/pull/1067)

>To make this easier for ROS users, I propose backporting whatever we need to humble to expose an ompl::ompl target.
Then, you can just do this instead:
find_package(ompl CONFIG REQUIRED)
add_executable(main main.cpp)
target_link_libraries(main PRIVATE ompl::ompl )

This is exactly what I am trying to do (when building both from source on the latest main in ROS Jazzy), but can't figure out how to make it work correctly.
`target_link_libraries(moveit_ompl_interface ompl::ompl)`

I even tried a source install of OMPL but still fail to find the target `ompl::ompl` so I suspect OMPL is missing an export somewhere.

Ryanf55, 2024-09-08:
Yea, for sure.  This all makes sense. I'm happy to work with the maintainers on implementing a solution, just need some buy-in on my proposal. There may be some less risky patches we could try, but I'd rather just spend the time to update humble (because that's what I use for all my development).

MarqRazz, 2024-09-09:
I was discussing this issue with @JafarAbdi and he was able to help me pinpoint the issue down to the location of the library being installed in the wrong location. I went to make a PR today and it looks like @mamoll and @v4hn made a [patch over the weekend](https://github.com/ompl/ompl/commit/e469b4d87809c431b297588d37782b0188a079b4) that can close this issue on the OMPL side. I will make a PR for MoveIt to update.

@mamoll, @JafarAbdi and myself would be happy to take over the release of OMPL. Please let us know if there is anything we can do to help.

v4hn, 2024-09-09:
What exactly needs patching on the moveit side to address this?

MarqRazz, 2024-09-09:
[Here is the PR](https://github.com/moveit/moveit2/pull/2994) to MoveIt2.

closed 2024-09-09 by MarqRazz