ompl/ompl #1176: [omplapp] Use external dependency ompl via CMake target instead of embedding

WangWeiLin-MV, 2024-08-05:
Hi,

I am a member of the VCPKG project. I found that `omplapp` repeatedly builds "ompl" from the submodule instead of depends on the built libraries https://github.com/microsoft/vcpkg/pull/40151.

I see that #1066 #1067 adds the CMake targets `ompl::ompl`, which allow for convenient `target_link_libraries(main PRIVATE ompl::ompl)`, but omplapp not uses it, and it bypasses `ompl/CMakeLists.txt` to compile its components like this [omplapp/CMakeLists.txt#L252-L257](https://github.com/ompl/omplapp/blob/1.6.0/CMakeLists.txt#L252-L257):
```cmake
add_subdirectory(ompl/doc)
add_subdirectory(ompl/src)
add_subdirectory(ompl/py-bindings)
add_subdirectory(ompl/tests)
add_subdirectory(ompl/demos)
add_subdirectory(ompl/scripts)
```

And there are many different parts between the two files [omplapp/CMakeLists.txt](https://github.com/ompl/omplapp/blob/1.6.0/CMakeLists.txt) and [ompl/CMakeLists.txt](https://github.com/ompl/ompl/blob/1.6.0/CMakeLists.txt). Splitting the two projects need a lot of changes.

I would like to ask if it is possible to modify omplapp to be a project that depends on ompl, rather than an ompl variant that shares a lot of code? This is very helpful to us, and I think this will also make it easier to develop `omplapp`.

If you have any questions please let me know. Thanks in advance.

Ryanf55, 2024-09-08:
Yes, agree. It will be easier after this PR.
https://github.com/ompl/omplapp/pull/13

If `omplapp` called `find_package(ompl CONFIG)` before it brought it in as a submodule, would that fix it for you?

WangWeiLin-MV, 2024-09-09:
> If `omplapp` called `find_package(ompl CONFIG)` before it brought it in as a submodule, would that fix it for you?

Yes, the duplicate build could be avoided in that way.

Ryanf55, 2024-09-09:
@mamoll  Can you move this to omplapp? This issue is specific to omplapp