ompl/ompl #1189: Planning a trajectory passing through some waypoints

michelemarrone, 2024-09-30:
Hi,
I'm working on a motion planning problem using sampling-based techniques in OMPL. The scenario involves:

1. A 6 degree-of-freedom system
2. Finding an optimal trajectory
3. Starting from an initial point
4. Visiting n other points in an optimal order
5. Not returning to the starting point

The key aspects of the problem are:

- Decision variable: The order in which to visit the n points
- Cost function: A metric based on the cost of connecting any two of the n nodes

I know it's similar to some kind of Travelling Salesmen Problem but not precisely.

Currently, my solution has been imposing the constraint to pass through the n points as an optimization objective for the algorithms but it's not giving me results.
Have you an advice for me?

Michele

michelemarrone, 2024-10-07:
Hi @kylc, thank you for the ideas, I will look into them.

Right now, I was trying to go in other direction and I hope you can tell me if it can be a feasible idea.
If I have a starting point, then I can use goalStates and find the closest one. Then starting from the closest point I can iterate by erasing the visited point from the goalStates until I visited all the points.

What do you think about it?

kylc, 2024-10-08:
It sounds like you are describing a greedy algorithm which always plans from the current state to the nearest one, and so on. If that's the case, then you are not really solving for the globally optimal path through all of the waypoints -- only the greedy approximation of such a path.

This simplified problem can easily be solved by just planning segments from each state to the nearest neighbor until the goal is reached, which is indeed a common approach and is often good enough.

michelemarrone, 2024-10-13:
Thank you. I see your point and I hope it is good enough for me.

closed 2024-10-13 by michelemarrone