ompl/ompl #1196: Python Bindings Build Failure on Debian (via WSL)

philipp1604, 2024-11-15:
### Python Bindings Build Failure on Debian (via WSL)

#### Issue:
1. **Initialization**:  
   - The build folder was initialized and the execution of `make update_bindings` enabled.
   - During configuration, the optional module `boost_python-py311` was reported as not found, despite the matching `libboost-python` being installed.

2. **Error during `make update_bindings`**:  
   - Errors occurred while creating C++ code for the Python module `util` (see `pyplusplus_util.log`).
   - **Warning `W1008`**: Non-const references (e.g., `std::vector<double>::reference`) to Python immutable types cannot be modified in Python.
   - **Compilation Error `W1050`**: Missing call policies for functions returning `std::vector<double>::pointer`, causing binding generation to fail.
  
### Attachments
[01_Pre_Build_Process.txt](https://github.com/user-attachments/files/17777104/01_Pre_Build_Process.txt)
[02_cmake_call.txt](https://github.com/user-attachments/files/17777105/02_cmake_call.txt)
[03_Building_Bindings_Error.txt](https://github.com/user-attachments/files/17777106/03_Building_Bindings_Error.txt)
[04_pyplusplus_util.txt](https://github.com/user-attachments/files/17777107/04_pyplusplus_util.txt)

#### Relevant Logs:

```bash
-- The CXX compiler identification is GNU 12.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.58") found components: serialization filesystem system program_options
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1")
-- Checking for module 'libspot'
--   Package 'libspot', required by 'virtual:world', not found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Found PY_numpy: /usr/lib/python3/dist-packages/numpy
-- Found PY_pyplusplus: /usr/local/lib/python3.11/dist-packages/pyplusplus
-- Found PY_pygccxml: /usr/local/lib/python3.11/dist-packages/pygccxml
-- Found castxml: /usr/bin/castxml
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * boost_numpy (required version == 1.74.0)
 * boost_python (required version == 1.74.0)
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * boost_unit_test_framework (required version == 1.74.0)
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.
 * Python, <https://www.python.org>
   Used for python bindings.

-- The following REQUIRED packages have been found:

 * boost_serialization (required version == 1.74.0)
 * boost_filesystem (required version == 1.74.0)
 * boost_system (required version == 1.74.0)
 * boost_program_options (required version == 1.74.0)
 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * boost_headers (required version == 1.74.0)
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.9.2), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).
 * boost_python-py311 (required version == 1.74.0)

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/phil/ompl/build/Release

- Build files have been written to: /home/phil/ompl/build/Release
[  0%] Preparing C++ header file for Python binding generation for module tools
[ 50%] Preparing C++ header file for Python binding generation for module util
[ 50%] Preparing C++ header file for Python binding generation for module control
[ 50%] Preparing C++ header file for Python binding generation for module base
[ 50%] Preparing C++ header file for Python binding generation for module geometric
[ 50%] Built target tools.h
[ 50%] Built target control.h
[ 50%] Built target base.h
[ 50%] Built target util.h
[ 50%] Built target geometric.h
[ 50%] Creating C++ code for Python module util (see pyplusplus_util.log)
/usr/local/lib/python3.11/dist-packages/pyplusplus/binary_parsers/parsers.py:7: DeprecationWarning: invalid escape sequence 'm'
  """
std::vector<double>::reference std::vector<double>::at(std::size_t
__n) [member function] > warning W1008: The function returns non-const
reference to "Python immutable" > type. The value cannot be modified
from Python.
std::vector<double>::pointer
std::vector<double>::_M_allocate_and_copy(std::size_t __n,
__gnu_cxx::__normal_iterator<const double *, std::vector<double>>
__first, __gnu_cxx::__normal_iterator<const double *,
std::vector<double>> __last) [member function] > compilation error
W1050: The function returns "std::vector<double>::pointer" > type. You
have to specify a call policies. Be sure to take a look on `Py++`
defined call policies.
static std::vector<double>::pointer
std::vector<double>::_S_do_relocate(std::vector<double>::pointer
__first, std::vector<double>::pointer __last,
std::vector<double>::pointer __result,
std::vector<double>::_Tp_alloc_type & __alloc, std::true_type arg4)
[member function] > compilation error W1050: The function returns
"std::vector<double>::pointer" > type. You have to specify a call
policies. Be sure to take a look on `Py++` > defined call policies.
static std::vector<double>::pointer
std::vector<double>::_S_do_relocate(std::vector<double>::pointer arg0,
std::vector<double>::pointer arg1, std::vector<double>::pointer
__result, std::vector<double>::_Tp_alloc_type & arg3, std::false_type
arg4) [member function] > compilation error W1050: The function
returns "std::vector<double>::pointer" > type. You have to specify a
call policies. Be sure to take a look on `Py++` > defined call policies.

kylc, 2024-11-15:
Are you able to use the prebuilt Python wheels available at https://github.com/ompl/ompl/releases/tag/prerelease? This removes all the complexity of building OMPL yourself.

If not, make sure you are installing the latest pyplusplus from https://github.com/ompl/pyplusplus. From your log, it looks like the install script is pulling an older version from PyPI.

philipp1604, 2024-12-03:
Thank you for your suggestion @kylc ! Installing the latest pyplusplus version did not resolve the issue. However, the prebuilt bindings you mentioned seem to work well on my system. I appreciate your help!

philipp1604, 2025-01-31:
@kylc I hope you're doing well! I wanted to follow up regarding the prebuilt Python wheels. Unfortunately, when attempting to run `ConstrainedPlanningTorus.py` with the pre-release version from [this link](https://github.com/ompl/ompl/releases/tag/prerelease), I encountered the following error:

`Exception has occurred: TypeError No Python class registered for C++ class Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> > File "/home/phil/projects/ompl/demos/constraint/ConstrainedPlanningCommon.py", line 215, in solveOnce stat = self.ss.solve(self.options.time) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/home/phil/projects/ompl/demos/constraint/ConstrainedPlanningTorus.py", line 140, in torusPlanningOnce stat = cp.solveOnce(output, "torus") ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/home/phil/projects/ompl/demos/constraint/ConstrainedPlanningTorus.py", line 185, in torusPlanning torusPlanningOnce(cp, planners[0], options.output) File "/home/phil/projects/ompl/demos/constraint/ConstrainedPlanningTorus.py", line 210, in <module> torusPlanning(parser.parse_args()) TypeError: No Python class registered for C++ class Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> >`

mamoll, 2025-04-11:
Can you try again with the latest wheels that you can download for the 1.7.0 release?