ompl/ompl #1202: Valid start state becomes invalid

cWonp, 2024-11-29:
Hi, I'm working on a very simple tutorial using RRT. 
Below is the code I'm using. 
The problem is that, even though the start state is valid, I keep getting an error saying the start state is invalid. Interestingly, if I constrain the quaternion of the start state to only have integer values, the error doesn't occur. Furthermore, when I set the goal state to be the same as the start state, there’s no "invalid" error, which suggests that the state itself isn’t truly invalid. Is there possibly a bound specific to the start state that I might have overlooked?
`
#define BOUND_X_LOW	-1000
#define BOUND_X_HIGH 1000
#define BOUND_Y_LOW -300
#define BOUND_Y_HIGH 300
#define BOUND_Z_LOW -800
#define BOUND_Z_HIGH 800

bool pathPlanner::initialize() {
	// Initialize stateSpace....................................
	space = std::make_shared<ob::SE3StateSpace>();

	// Set bound of state space (Limitation the search space)
	ob::RealVectorBounds bounds(3);
	// x axis
	bounds.setLow(0, BOUND_X_LOW);
	bounds.setHigh(0, BOUND_X_HIGH);
	// y axis
	bounds.setLow(1, BOUND_Y_LOW);
	bounds.setHigh(1, BOUND_Y_HIGH);
	// z axis 
	bounds.setLow(2, BOUND_Z_LOW);
	bounds.setHigh(2, BOUND_Z_HIGH);

	space->setBounds(bounds);

	// Initialize SimpleSetup....................................
	ss = std::make_shared<og::SimpleSetup>(space);

	// Set state validity checker
	ss->setStateValidityChecker([this](const ob::State * state) { return this->isStateValid(state); });
	
}

bool pathPlanner::isStateValid(const ob::State* state) {
	return true;
}

bool pathPlanner::getPath(m4f startPose, m4f goalPose) {
	// Set start and goal state
	ob::ScopedState<ob::SE3StateSpace> start(space);
	ob::ScopedState<ob::SE3StateSpace> goal(space);

	// Convert input tf -> quternion
	Eigen::Quaternionf q_start(startPose.block<3, 3>(0, 0));
	Eigen::Quaternionf q_goal(goalPose.block<3, 3>(0, 0));
	q_start.normalize();
	q_goal.normalize();

	// Set start and goal state
	start->setXYZ(startPose(0, 3), startPose(1, 3), startPose(2, 3));
	std::cout << start->getX() << ", " << start->getY() << ", " << start->getZ() << std::endl;
	start->rotation().setIdentity();
	start->rotation().w = q_start.w();
	start->rotation().x = q_start.x();
	start->rotation().y = q_start.y();
	start->rotation().z = q_start.z();
	

	goal->setXYZ(goalPose(0, 3), goalPose(1, 3), goalPose(2, 3));
	std::cout << goal->getX() << ", " << goal->getY() << ", " << goal->getZ() << std::endl;
	goal->rotation().setIdentity();
	goal->rotation().w = q_goal.w();
	goal->rotation().x = q_goal.x();
	goal->rotation().y = q_goal.y();
	goal->rotation().z = q_goal.z();

	ss->setStartAndGoalStates(goal, goal);

	// Set planner
	auto planner(std::make_shared<og::RRT>(ss->getSpaceInformation()));
	ss->setPlanner(planner);

	ob::PlannerStatus solved = ss->solve(1.0); // Specified amount of time (sec)

	bool res = false;
	if (solved) {
		std::cout << "Found solution!" << std::endl;
		ss->simplifySolution();
		ss->getSolutionPath().print(std::cout);
		res = true;
	}
	else {
		std::cout << "No solution found" << std::endl;
	}
	
	return res;
}`
'

the input and error msg is below:
`start XYZ: 785.97, 0.37, 300.05
start rotation:
w 0.00872656, x 0.999962, y 0, z 0
goal XYZ: 810, 0.37, 300.05
goal rotation:
w 0.544641, x -0.451445, y -0.448387, z 0.546363
[RBdata] TCP comm connect done
Debug:   RRT: Planner range detected to be 526.432017
Warning: RRT: Skipping invalid start state (invalid bounds)
         at line 249 in C:\Users\owner\Documents\roboticScanWs\vcpkgManage\vcpkg\buildtrees\ompl\src\1.6.0-33eb9d7aae.clean\src\ompl\base\src\Planner.cpp
Debug:   RRT: Discarded start state Compound state [
RealVectorState [785.97 0.37 300.05]
SO3State [0.999962 0 0 0.00872656]
]

Error:   RRT: There are no valid initial states!
         at line 112 in C:\Users\owner\Documents\roboticScanWs\vcpkgManage\vcpkg\buildtrees\ompl\src\1.6.0-33eb9d7aae.clean\src\ompl\geometric\planners\rrt\src\RRT.cpp
Info:    No solution found after 0.002012 seconds
No solution found
`

kylc, 2024-12-01:
Are the values you've printed to the full numerical precision used in the code? If so, the start state quaternion magnitude exceeds the unit quaternion tolerance in OMPL and so is out of bounds.

See https://github.com/ompl/ompl/blob/main/src/ompl/base/spaces/src/SO3StateSpace.cpp#L47

mamoll, 2025-03-05:
no further feedback from user, closing this issue

closed 2025-03-05 by mamoll