ompl/ompl #1203: State Overwriting in MotionValidator

michelemarrone, 2024-12-02:
Hi, I have encountered an issue with managing generic motions between two states, s1 and s2. The problem arises when s1 satisfies a specific condition that restricts control actions within a certain area of the configuration space. Here's a detailed explanation:

I attempt to perform a control action in s1 to reach s2.
However, if s1 lies within a non-maneuvering area (a specific region where control actions are not allowed), it is considered invalid as a maneuvering point. Conversely, states outside this area are valid.
The challenge is that s1 originates as the previous s2. If s2 is in the non-maneuvering area, the subsequent motion would be invalid by default.

To resolve this, I adjust the logic as follows:

When a state s1 lies within the non-maneuvering area, I identify the closest valid state outside this area.
In the MotionValidator, if the current s1 is in the non-maneuvering area, I reassign it (overwrite it) to this closest valid point.

Here is a snippet of the code implementing this:

```
s1->as<ompl::base::RealVectorStateSpace::StateType>()->values[j] = propagated_state[j];
s1->as<ompl::base::RealVectorStateSpace::StateType>()->values[6] = t + dt;
```
with a simple pseudocode:

```
function checkMotion(s1, s2):
    for each non-maneuvering area:
        if s1 is inside the non-maneuvering area:
                propagate s1 to the closest valid state
                if propagation succeed:
                    update s1 to the new state
                    true
                else
                    a close valid state does not exist
                    return false
    return true
```
Questions:
1) Is this approach correct?
2) Is there a better or more efficient way to handle this scenario?

Thank you!

kylc, 2024-12-03:
I doubt that it's safe to modify the states inside the `checkMotion` function like you suggest, if it's even possible (the states are `const`).

It sounds like you are trying to manually implement something which already exists in OMPL. See [Constrained Planning](https://ompl.kavrakilab.org/constrainedPlanning.html).

michelemarrone, 2024-12-24:
Thank you!
It was honestly a bit weird, I will look into it.