ompl/ompl #1209: Help

NXZXH, 2024-12-24:
(Genesis) robot@robot:~/下载$ pip install ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl
ERROR: ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl is not a supported wheel on this platform.
(Genesis) robot@robot:~/下载$ pip install "/home/robot/下载/ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl"
ERROR: ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl is not a supported wheel on this platform.
(Genesis) robot@robot:~/下载$ uname -m
x86_64
(Genesis) robot@robot:~/下载$ python --version
Python 3.10.16
(Genesis) robot@robot:~/下载$

kylc, 2025-02-04:
`pip debug` will show compatible tags for your platform. If it's not the architecture nor Python version, it may be the glibc version.

milong26, 2025-02-18:
> `pip debug` will show compatible tags for your platform. If it's not the architecture nor Python version, it may be the glibc version.

My `pip debug --verbose` outpus  is
<details>

WARNING: This command is only meant for debugging. Do not use this with automation for parsing and getting these details, since the output and options of this command may change without notice.

pip version: pip 25.0 from .../python3.10/site-packages/pip (python 3.10)
sys.version: 3.10.0 (default, Mar  3 2022, 09:58:08) [GCC 7.5.0]
sys.executable: /data/zly/application/anaconda3/envs/genesis/bin/python
sys.getdefaultencoding: utf-8
sys.getfilesystemencoding: utf-8
locale.getpreferredencoding: UTF-8
sys.platform: linux
sys.implementation:
  name: cpython
'cert' config value: Not specified
REQUESTS_CA_BUNDLE: None
CURL_CA_BUNDLE: None
pip._vendor.certifi.where(): /data/zly/application/anaconda3/envs/genesis/lib/python3.10/site-packages/pip/_vendor/certifi/cacert.pem
pip._vendor.DEBUNDLED: False
vendored library versions:
  CacheControl==0.14.1
  distlib==0.3.9
  distro==1.9.0
  msgpack==1.1.0
  packaging==24.2
  platformdirs==4.3.6
  pyproject-hooks==1.2.0
  requests==2.32.3
  certifi==2024.08.30
  idna==3.10
  urllib3==1.26.20
  rich==13.9.4 (Unable to locate actual module version, using vendor.txt specified version)
  pygments==2.18.0
  typing_extensions==4.12.2 (Unable to locate actual module version, using vendor.txt specified version)
  resolvelib==1.0.1
  setuptools==70.3.0 (Unable to locate actual module version, using vendor.txt specified version)
  tomli==2.2.1
  truststore==0.10.0
Compatible tags: 634
  cp310-cp310-manylinux_2_27_x86_64
  cp310-cp310-manylinux_2_26_x86_64
  cp310-cp310-manylinux_2_25_x86_64
  cp310-cp310-manylinux_2_24_x86_64
  cp310-cp310-manylinux_2_23_x86_64
  cp310-cp310-manylinux_2_22_x86_64
  cp310-cp310-manylinux_2_21_x86_64
  cp310-cp310-manylinux_2_20_x86_64
  cp310-cp310-manylinux_2_19_x86_64
  cp310-cp310-manylinux_2_18_x86_64
  cp310-cp310-manylinux_2_17_x86_64
  cp310-cp310-manylinux2014_x86_64
  cp310-cp310-manylinux_2_16_x86_64
  cp310-cp310-manylinux_2_15_x86_64
  cp310-cp310-manylinux_2_14_x86_64
  cp310-cp310-manylinux_2_13_x86_64
  cp310-cp310-manylinux_2_12_x86_64
  cp310-cp310-manylinux2010_x86_64
  cp310-cp310-manylinux_2_11_x86_64
  cp310-cp310-manylinux_2_10_x86_64
  cp310-cp310-manylinux_2_9_x86_64
  cp310-cp310-manylinux_2_8_x86_64
  cp310-cp310-manylinux_2_7_x86_64
  cp310-cp310-manylinux_2_6_x86_64
  cp310-cp310-manylinux_2_5_x86_64
  cp310-cp310-manylinux1_x86_64
  cp310-cp310-linux_x86_64
  cp310-abi3-manylinux_2_27_x86_64
  cp310-abi3-manylinux_2_26_x86_64
[... 583 lines cut ...]
  py30-none-manylinux_2_11_x86_64
  py30-none-manylinux_2_10_x86_64
  py30-none-manylinux_2_9_x86_64
  py30-none-manylinux_2_8_x86_64
  py30-none-manylinux_2_7_x86_64
  py30-none-manylinux_2_6_x86_64
  py30-none-manylinux_2_5_x86_64
  py30-none-manylinux1_x86_64
  py30-none-linux_x86_64
  cp310-none-any
  py310-none-any
  py3-none-any
  py39-none-any
  py38-none-any
  py37-none-any
  py36-none-any
  py35-none-any
  py34-none-any
  py33-none-any
  py32-none-any
  py31-none-any
  py30-none-any
</details>

How to do with this? Thanks!

kylc, 2025-02-18:
OMPL wheels are built only for glibc >=2.28 (the `2_28` in the wheel file name). Your system is only compatible with glibc <=2.27. You should upgrade your system, or build the wheels from source.

milong26, 2025-02-19:
> OMPL wheels are built only for glibc >=2.28 (the `2_28` in the wheel file name). Your system is only compatible with glibc <=2.27. You should upgrade your system, or build the wheels from source.

Does `build the wheels from source` mean install following this [installation](https://ompl.kavrakilab.org/core/installation.html)? I tried but it needs sudo，not suitable for me xD

Thanks!

closed 2025-10-02 by saahu27