ompl/ompl #1211: Solver throws a validation failure for infinite bound variables

alexis779, 2025-01-02:
### Context

I ran into this issue when running [motion planning](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/inverse_kinematics_motion_planning.html) in genesis-world library. The joint variables in the model were unbounded.

Motion planning should support the unbounded use case right?

- I could not figure out how to run python binding locally from source. Running into this [issue](https://github.com/ompl/ompl/issues/1110).
- I am not sure how to run the python debugger and step through the C++ library. The binding would probably need to be rebuilt with Debug flag enabled ...
- My next step would be to try to reproduce this as a C++ unit test.

### Error

When passing infinite boundary, the solver fails with a validation error in dimension `>= 3`

```
======================================================================
ERROR: testInfiniteBound (__main__.MotionPlanningTest.testInfiniteBound)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/alexis/Documents/python/robotics/slobot/test_ompl_infinity_bounds.py", line 11, in testInfiniteBound
    self.solveOmpl(-np.inf, np.inf)
  File "/home/alexis/Documents/python/robotics/slobot/test_ompl_infinity_bounds.py", line 42, in solveOmpl
    self.assertTrue(ss.solve(timeout))
                    ^^^^^^^^^^^^^^^^^
RuntimeError: Bounds for real vector space seem to be incorrect (lower bound must be stricly less than upper bound). Sampling will not be possible
```

### Unit Test

```python
import unittest
from ompl import base as ob
from ompl import geometric as og
import numpy as np

class MotionPlanningTest(unittest.TestCase):
    def testFiniteBound(self):
        self.solveOmpl(-3.14, 3.14)

    def testInfiniteBound(self):
        self.solveOmpl(-np.inf, np.inf)

    def solveOmpl(self, lower, upper):
        n_qs = 3
        space = ob.RealVectorStateSpace(n_qs)
        bounds = ob.RealVectorBounds(n_qs)

        for i_q in range(n_qs):
            bounds.setLow(i_q, lower)
            bounds.setHigh(i_q, upper)

        space.setBounds(bounds)
        ss = og.SimpleSetup(space)
        ss.setStateValidityChecker(ob.StateValidityCheckerFn(lambda state: True))

        planner = "RRTConnect"
        ss.setPlanner(getattr(og, planner)(ss.getSpaceInformation()))

        state_start = ob.State(space)
        state_goal = ob.State(space)

        qpos_start = np.zeros(n_qs)
        qpos_goal = np.ones(n_qs)

        for i_q in range(n_qs):
            state_start[i_q] = float(qpos_start[i_q])
            state_goal[i_q] = float(qpos_goal[i_q])

        ss.setStartAndGoalStates(state_start, state_goal)

        timeout = 5
        self.assertTrue(ss.solve(timeout))

if __name__ == '__main__':
    unittest.main()

```

### Conda env

Store conda configuration as `environment.yml`
```
name: ompl
channels:
  - defaults
dependencies:
  - python=3.12.8
  - pip
  - pip:
    - https://github.com/ompl/ompl/releases/download/prerelease/ompl-1.6.0-cp312-cp312-manylinux_2_28_x86_64.whl
    - numpy<2
```

Create conda env

```
conda env create -f ./environment.yml
conda activate ompl
```

marip8, 2025-01-07:
FWIW, I ran into a case like this recently and here are my takeaways:

> Motion planning should support the unbounded use case right?

Since OMPL planners are based on discrete sampling with fixed precision (i.e. bounded) numbers, I don't think it's technically possible to support unbounded joint ranges (i.e. ranges that contain `inf`). If your joints are truly unbounded, I think the best you could do is to set their limits to the maximum float value and sample in that very large (but bounded) range instead.

> I am not sure how to run the python debugger and step through the C++ library. The binding would probably need to be rebuilt with Debug flag enabled ...

Correct. I was able to do this with the bindings that I built locally

alexis779, 2025-01-08:
Thanks for the feedback!

I had run into this error because the URDF configuration for a robotic arm I was using had [continuous joints](http://wiki.ros.org/urdf/XML/joint#Attributes). Like you said, as a workaround it makes sense to change them into revolute, with arbitrary large lower & upper limits, as long as they're finite.

closed 2025-01-08 by alexis779