ompl/ompl #1221: Can not create PRMstar with PlannerData through python binding.

Riften, 2025-01-15:
## Description

I'm trying to use PRMstar planner with reusable roadmap. As described in this issue https://github.com/ompl/ompl/issues/936 , I can do it by initializing PRMstar planner with given PlannerData. I can also find the constructor of PRMstar with PlannerData as argument in [source code](https://ompl.kavrakilab.org/PRMstar_8h_source.html).

But when I try it through python binding, I got error 

```bash
ArgumentError: Python argument types in
    PRMstar.__init__(PRMstar, PlannerData)
did not match C++ signature:
    __init__(_object*, std::shared_ptr<ompl::base::SpaceInformation> si)
```

## Environment
- Python 3.9
- OMPL 1.6.0

## Reproduce

You can reproduce it by following python code

```python
from ompl import base as ob
from ompl import geometric as og
space = ob.RealVectorStateSpace(2)
bounds = ob.RealVectorBounds(2)
bounds.setLow(-1)
bounds.setHigh(1)
space.setBounds(bounds)
simple_setup = og.SimpleSetup(space)
planner = og.PRMstar(simple_setup.getSpaceInformation())
start_state = ob.State(space)
goal_state = ob.State(space)
start_state.random()
goal_state.random()
simple_setup.setStartAndGoalStates(start_state, goal_state)
simple_setup.setPlanner(planner)
print(simple_setup.solve(0.1))
planner_data = ob.PlannerData(simple_setup.getSpaceInformation())
planner.getPlannerData(planner_data)
storage = ob.PlannerDataStorage()
storage.store(planner_data, "test_planner_data")

# read back the planner data
simple_setup.clear()
loaded_data = ob.PlannerData(simple_setup.getSpaceInformation())
storage.load("test_planner_data", loaded_data)
new_planner = og.PRMstar(loaded_data)
simple_setup.setPlanner(new_planner)
simple_setup.setStartAndGoalStates(start_state, goal_state)
print(simple_setup.solve(0.1))
```

<details>
<summary>Full output</summary>

```bash
Exact solution
Info:    PRMstar: Starting planning with 2 states already in datastructure
Info:    PRMstar: Created 608 states
Info:    Solution found in 0.100148 seconds
Warning: State validity checker not set! No collision checking is performed
         at line 63 in /project/src/ompl/base/src/SpaceInformation.cpp
---------------------------------------------------------------------------
ArgumentError                             Traceback (most recent call last)
Cell In[14], line 26
     24 loaded_data = ob.PlannerData(simple_setup.getSpaceInformation())
     25 storage.load("test_planner_data", loaded_data)
---> 26 new_planner = og.PRMstar(loaded_data)
     27 simple_setup.setPlanner(new_planner)
     28 simple_setup.setStartAndGoalStates(start_state, goal_state)

ArgumentError: Python argument types in
    PRMstar.__init__(PRMstar, PlannerData)
did not match C++ signature:
    __init__(_object*, std::shared_ptr<ompl::base::SpaceInformation> si)
```

</details>

However, there is no exception when using PRM planner instead of PRMstar.

<details>
<summary>Full output with PRM</summary>

```bash
Info:    PRM: Starting planning with 2 states already in datastructure
Exact solution
Info:    PRM: Created 880 states
Info:    Solution found in 0.100231 seconds
Info:    PRM: Starting planning with 884 states already in datastructure
Info:    PRM: Created 1451 states
Info:    Solution found in 0.100232 seconds
Exact solution
Warning: State validity checker not set! No collision checking is performed
         at line 63 in /project/src/ompl/base/src/SpaceInformation.cpp
```

</details>

kylc, 2025-04-07:
FYI for anyone who wants to dig deeper.

The cause is likely a missing constructor in the PRMstar Python wrapper of the form `PRMstar_wrapper(const PlannerDataPtr&)` here: https://github.com/ompl/ompl/blob/main/py-bindings/generate_bindings.py#L893-L897

mamoll, 2025-04-07:
Thanks, @kylc!
Fix in https://github.com/ompl/ompl/pull/1267

closed 2025-04-07 by Riften