ompl/ompl #1225: Modules compiled with NumPy 1.x are not directly compatible with NumPy 2.x.

kelejichi6463, 2025-01-25:
ubuntu22.04
python 3.10

$ python3
Python 3.10.12 (main, Jan 17 2025, 14:35:34) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ompl
>>> from ompl import base as ob

A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.1.3 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

Traceback (most recent call last):  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/ompl/base/__init__.py", line 2, in <module>
    from ompl.base._base import *
AttributeError: _ARRAY_API not found
ImportError: numpy.core._multiarray_umath failed to import
>>> from ompl import _base as ob
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name '_base' from 'ompl' (/usr/lib/python3/dist-packages/ompl/__init__.py)

kylc, 2025-02-04:
Please make sure you are testing with this recent commit: https://github.com/ompl/ompl/commit/5aa3ec0beae287d8de267b94cdb935e68f19d680

saahu27, 2025-03-13:
Numpy dependency has been updated and is available in the latest commit and also in the upcoming release. Hence closing the issue.

closed 2025-03-13 by saahu27