ompl/ompl #1228: Prebuilt Python Wheels from the Development Build Do Not Support Constrained Planning

philipp1604, 2025-02-01:
Constrained planning is generally not supported by the development build, as can be demonstrated by running, for example, the `ConstrainedPlanningTorus.py` file from the [Development Build](https://github.com/ompl/ompl/releases/tag/prerelease). I encountered the following error:

`Exception has occurred: TypeError No Python class registered for C++ class Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> > File "/home/phil/projects/ompl/demos/constraint/ConstrainedPlanningCommon.py", line 215, in solveOnce stat = self.ss.solve(self.options.time) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/home/phil/projects/ompl/demos/constraint/ConstrainedPlanningTorus.py", line 140, in torusPlanningOnce stat = cp.solveOnce(output, "torus") ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/home/phil/projects/ompl/demos/constraint/ConstrainedPlanningTorus.py", line 185, in torusPlanning torusPlanningOnce(cp, planners[0], options.output) File "/home/phil/projects/ompl/demos/constraint/ConstrainedPlanningTorus.py", line 210, in <module> torusPlanning(parser.parse_args()) TypeError: No Python class registered for C++ class Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> >`

Note: The rest of the Python examples outside of constrained planning work without any issues.

m-decoster, 2025-03-18:
I have the same issue, but not with all pre-built wheels. For example, in an old environment I can use constrained planning:

```
$ pip freeze | grep ompl
ompl @ https://github.com/ompl/ompl/releases/download/prerelease/ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl#sha256=71f3f056e67955c588ec602a5f145cc212113aac909257bd4dcdac31845c53dd
```

but in a newly created environment, where the wheel is different, I get the same error as @philipp1604 

```
$ pip freeze | grep ompl
ompl @ file:///home/mcdcoste/Downloads/ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl#sha256=deaf54cc167e078adf60491ca2a5d8b372cd6c209e71605da3f8bc441cb8799c
```

kylc, 2025-04-07:
@m-decoster The SHA sums of your OMPL wheels in the two environments do not match. Perhaps try testing only with the latest OMPL prerelease wheels. My guess it that your error is related to issue https://github.com/ompl/ompl/pull/1159 and is already fixed.

m-decoster, 2025-04-07:
@kylc the first SHA was an older wheel, the second SHA was the development build when I posted that comment 3 weeks ago, so it may be a regression from #1159.

kylc, 2025-04-07:
I'm not able to reproduce the issue in a minimal Docker environment. Perhaps you can narrow down how your environment differs from this one:
```
# on host
docker run -it --rm ubuntu:22.04

# in container
apt update && apt install -y python3-pip python3-numpy git
python3 -m pip install https://github.com/ompl/ompl/releases/download/prerelease/ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl
git clone https://github.com/ompl/ompl --depth 1
python3 ompl/demos/constraint/ConstrainedPlanningTorus.py
# =>
# Info:    Using Projection-Based State Space!
# ...
```

mamoll, 2025-04-07:
Can you check the 1.7.0 wheels and let me know if they work for you?

m-decoster, 2025-04-08:
@kylc Unfortunately, the download link https://github.com/ompl/ompl/releases/download/prerelease/ompl-1.6.0-cp310-cp310-manylinux_2_28_x86_64.whl is no longer available and results in a HTTP 404 error.

I tried your script with the suggestion of @mamoll to use 1.7.0 wheels, and this appears to work:

```
# Inside the container
apt update && apt install -y python3-pip python3-numpy git
apt install wget unzip
wget https://github.com/ompl/ompl/releases/download/1.7.0/wheels-ubuntu-latest-x86_64.zip
unzip wheels-ubuntu-latest-x86_64.zip 
pip install ompl-1.7.0-cp310-cp310-manylinux_2_28_x86_64.whl 
git clone https://github.com/ompl/ompl --depth 1
python3 ompl/demos/constraint/ConstrainedPlanningTorus.py
```

gives the output `Info:    Using Projection-Based State Space!` (and then some more)

This appears to be fixed.

closed 2025-04-11 by mamoll