ompl/ompl #1240: new release version 1.7.0

johnnynunez, 2025-03-05:
@mamoll Could you create new tag release version?
pygccxml fixes are done! Now, you can build for aarch, macos (x86,arm), linux(aarch,arm), windows 2025 with castxml 0.6.11

mamoll, 2025-03-05:
we're working on it. the ompl web page needs a bit of refresh before the next release.

johnnynunez, 2025-03-05:
> we're working on it. the ompl web page needs a bit of refresh before the next release.

thanks a lot!

mamoll, 2025-04-07:
Release is out, so closing this issue

johnnynunez, 2025-04-07:
the name of the wheels are wrong. Also, will it be published on pypi?

[image]

mamoll, 2025-04-08:
That's an oversight. Fixing it now. Rebuilding the wheels takes a while. Thanks for letting us know.