ompl/ompl #1262: OMPL PYBINDINGS are working in Debian?

lepalom, 2025-04-03:
Hi,

I'm trying to build ompl pybindings in a Debian 12. I have following the instructions of Ubuntu, but I have done it in a virtual environment.
```
$ pip3 list
Package    Version
---------- -------
pip        23.0.1
pygccxml   3.0.2
pyplusplus 1.8.7
setuptools 66.1.1
```

However, when I run cmake, I got:
```
-- The following Python modules were NOT found:

 * pyplusplus

``` 

Also, I would like to ask which dependencies (and versions) has ompl to build pybindings? And, once ompl pybindings are built, the dependencies are still dependencies or can be dropped?

mamoll, 2025-04-03:
Make sure the right version of python is used. You may have to remove the CMakeCache.txt in your build directory and rerun cmake like so:
```
cmake -DPYTHON_EXEC=/usr/bin/python3.x ..
```
where `3.x` is the version for which you installed pyplusplus.

lepalom, 2025-04-04:
@mamoll I got the same:

```
 cmake -DOMPL_REGISTRATION=OFF -DPYTHON_EXEC=/usr/bin/python3.11 ..
...

-- The following Python modules were NOT found:

 * pyplusplus
```

maybe it is because I'm working in a python venv?

kylc, 2025-04-07:
@lepalom If you are working in a virtualenv then you should not provide the system python path. Use `which python3` while the virtualenv is active to find the correct `PYTHON_EXEC` path.

For example:
```
$ . venv/bin/activate
$ which python3
/ompl/venv/bin/python
$ python3 -c "import pygccxml; print(pygccxml)"
<module 'pygccxml' from '/ompl/venv/lib/python3.13/site-packages/pygccxml/__init__.py'>
$ cmake -DPYTHON_EXEC=$(which python3) ..
```

lepalom, 2025-04-07:
yes, now it works. But once ompl pybindings are built, the dependencies are still dependencies or can be dropped?

mamoll, 2025-04-08:
pygccxml and pyplusplus are only needed to generate the bindings. Once the bindings are built, they are no longer needed. You can close this issue if that answers your concerns.

closed 2025-04-08 by lepalom