ompl/ompl #1270: Possible memory corruption in ompl-1.7 bindings

srmainwaring, 2025-04-08:
There may be an issue causing memory corruption in the latest binding library. Still investigating, but the issue occurs when using `pytest` to run Python test cases.

- Does occur on an Ubuntu 24.04 docker image using the published wheels.
- Does not occur on macOS when the bindings are built from source against Python 3.13/

## Details

Minimal example to replicate. Create a test case that imports the library:

```python
# test_ompl_import.py
def test_ompl_import()
    from ompl import base as ob

```

Run the test:

```bash
$ pytest ./test_ompl_import.py 
=============================================================== test session starts ===============================================================
platform linux -- Python 3.10.12, pytest-8.3.5, pluggy-1.5.0
rootdir: /terrain_nav_py
configfile: pyproject.toml
collected 1 item                                                                                                                                  

test_ompl_import.py .                                                                                                                       [100%]

================================================================ 1 passed in 0.14s ================================================================
corrupted size vs. prev_size while consolidating
Aborted
```

kylc, 2025-04-09:
I'm running into this as well. gdb complains of some illegal memory access of type `std::vector<DubinsPathSegmentType>`, which pointed me to this commit: https://github.com/ompl/ompl/commit/2e953fe8c620b72cbb87d3b90ee635c236371215. Dropping this commit only (everything else stays the same) causes the test to succeed, while adding it back in causes the test to fail.

Perhaps something about the RAII free ordering of the static vectors vs. the pointers into them?

Relevant stack trace:
```
#7  0x00007ffff66e831c in std::__new_allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType>::deallocate (this=0x49b970, __n=<optimized out>, 
    __p=<optimized out>) at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/new_allocator.h:172
#8  std::allocator_traits<std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> >::deallocate (__n=<optimized out>, __p=<optimized out>, 
    __a=...) at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/alloc_traits.h:517
#9  std::_Vector_base<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> >::_M_deallocate (__n=<optimized out>, __p=<optimized out>, this=0x49b970)
    at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/stl_vector.h:390
#10 std::_Vector_base<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> >::~_Vector_base (this=0x49b970, __in_chrg=<optimized out>) at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/stl_vector.h:369
#11 std::vector<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> >::~vector (
    this=0x49b970, __in_chrg=<optimized out>) at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/stl_vector.h:738
#12 std::_Destroy<std::vector<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> > > (
    __pointer=0x49b970) at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/stl_construct.h:151
#13 std::_Destroy_aux<false>::__destroy<std::vector<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> >*> (__last=0x49ba00, __first=0x49b970) at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/stl_construct.h:163
#14 std::_Destroy<std::vector<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> >*> (
    __last=0x49ba00, __first=<optimized out>) at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/stl_construct.h:196
#15 std::_Destroy<std::vector<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> >*, std::vector<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> > > (__last=0x49ba00, 
    __first=<optimized out>) at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/alloc_traits.h:948
#16 std::vector<std::vector<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> >, std::allocator<std::vector<ompl::base::DubinsStateSpace::DubinsPathSegmentType, std::allocator<ompl::base::DubinsStateSpace::DubinsPathSegmentType> > > >::~vector
    (this=0x7ffff6c10190 <ompl::base::DubinsStateSpace::dubinsPathType>, __in_chrg=<optimized out>)
    at /nix/store/yg4ahy7gahx91nq80achmzilrjyv0scj-gcc-13.3.0/include/c++/13.3.0/bits/stl_vector.h:735
```

srmainwaring, 2025-04-09:
@kylc - thanks for the stack trace. The change was introduced to allow bindings to be generated for the `PathType`, but has introduced a static initialisation / destruction bug. Previously allocation of the c-style array was at compile time, now its at run time so there is an ordering problem.

@mamoll there is a nice article on the issue here: https://ppwwyyxx.com/blog/2023/Safe-Static-Initialization-No-Destruction. 

There are a couple choices:

1. Revert the change and solve the problem of generating and registering binding for c-style arrays.
2. Use one of the approaches outlined in the article to control initialisation and use one of the safe destruction approaches mentioned.  

I'd like to keep the bindings, so will prepare a patch for 2.

closed 2025-04-14 by mamoll