ompl/ompl #1277: how to install ompl in a anaconda env

gyq-china, 2025-05-02:
hello，i am a student，I want to install ompl-py in a virtual environment
I installed it according to the official script, but it seems to be installed globally and cannot be used in the virtual environment. Then I searched some blogs to modify the content in the script to use the py interpreter boost in the anaconda environment, etc., but all failed.

saahu27, 2025-05-02:
hi @gyq-china -> ompl is now available on PyPi. You can run pip install ompl inside the conda env.

gyq-china, 2025-05-03:
> hi [@gyq-china](https://github.com/gyq-china) -> ompl is now available on PyPi. You can run pip install ompl inside the conda env.

Thank you so much！！

closed 2025-05-03 by gyq-china