ompl/ompl #1284: No pypi wheels for Windows/no working conda-forge version for Python bindings?

david-dorf, 2025-05-31:
Appears to be no current wheels for Win64 here: https://pypi.org/project/ompl/#files

Separately, conda-forge ompl (https://anaconda.org/conda-forge/ompl) could not be used in Python (ie. `import ompl`).

saahu27, 2025-06-01:
@david-dorf Unfortunately, we dont have python wheels for windows. We are actively working towards developing bindings for crossplatform.

mamoll, 2025-06-03:
@david-dorf none of the core OMPL developers develop on Windows, so we depend on external contributors for this kind of improvement. The build process for python bindings is a bit involved. Our CI system uses [this pipeline](https://github.com/ompl/ompl/blob/main/.github/workflows/wheels.yaml) and [this setup.py](https://github.com/ompl/ompl/blob/main/py-bindings/setup.py) to build wheels. If you want just a Windows build of the C++ library, you could also look at the vcpkg build recipe and use that as a starting point. 

Having said all that, I wouldn't put too much effort into it as we are in the process of a major overhaul of the Python bindings. See PR  #1278 (note that this PR is against a development branch, not `main`, since there is still quite a bit of work left before it's ready to be merged into main).

closed 2025-10-01 by saahu27