ompl/ompl #1289: Contribution of BLIT* Algorithm to OMPL

yi213-robotic, 2025-06-07:
Dear OMPL Team,
     I am Yi Wang, and I am currently collaborating with Professor Oren Salzman from Technion. We are excited to share that our algorithm, Bidirectional Lazy Informed trees (BLIT*), has been accepted for presentation at ICRA 2025. 

     BLIT* is the first algorithm that incorporates anytime incremental lazy bidirectional heuristic search into batch-wise sampling-based motion planning (Bw-SBMP). Specifically, BLIT* introduces the first anytime incremental lazy heuristic search at the search layer and also develops a new lazy edge evaluation strategy. 

     We believe this work could meaningfully extend OMPL’s capabilities, and we are reaching out to explore the possibility of contributing it to the OMPL library.

     Attached are illustrations showing how motion trees for evaluated planners, as well as experimental results presented in our ICRA 2025 oral talk. These results span different dimensional spaces and robotic systems.

     If our contribution aligns with OMPL's direction and standards, we would be grateful for your guidance on how to proceed with the submission process.

    Thank you for your time and consideration. We look forward to your feedback.

[image]

[image]

Thank you,

Yi

saahu27, 2025-06-08:
Hi @yi213-robotic , Thank you for your contribution. I coudn't find the Paper in arxiv. Could you please help me find the Paper.

yi213-robotic, 2025-06-08:
> Hi [@yi213-robotic](https://github.com/yi213-robotic) , Thank you for your contribution. I coudn't find the Paper in arxiv. Could you please help me find the Paper.

closed 2025-06-08 by yi213-robotic

saahu27, 2025-06-08:
@yi213-robotic Thank You for the presentation slides. Great Work. 
Please make a pull request to the main ompl if you would like to make a contribution, and the team would review it and respond.