ompl/ompl #1301: Attempting to build on Mac OSX : "No rule to make target `update_bindings'" - bindings not installed!?

DXXS, 2025-07-27:
['ompl' appears to be a ~newly introduced dependency of 'genesis-world' main branch {'requirement ompl>=1.7.0'}, which, after 'brew' indicated that they don't install the python bindings, led me to attempt to build]:

After following your build instructions, attempted the recommended command to build with python bindings

   Release % make -j 4 update_bindings
   make: *** No rule to make target `update_bindings'.  Stop.

I noticed that in the "cmake ../.." command, the output included:
   "-- The following features have been disabled:
    * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings."

So I attempted to "cmake ../.. -DOMPL_BUILD_PYBINDINGS=ON" instead, as apparently recommended by other of your issues, which removed that notice...

[FWIW Later calls 'make install' did appear to install some python example code!?]

However, 'update_bindings ' still not recognized; in grep searching the Release folder, only "install-ompl-ubuntu.sh" appears to use it, in a manner not supported by the cmake on my system!?

The 'Makefile' shows there is a target 'clean_bindings', but that doesn't seem to do the trick either...

::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::

Release % cmake --version            
cmake version 3.23.0

DXXS, 2025-07-27:
Release % python /usr/local/share/ompl/demos/OptimalPlanning.py

Traceback (most recent call last):
  File "/usr/local/share/ompl/demos/OptimalPlanning.py", line 41, in <module>
    from ompl import util as ou
ModuleNotFoundError: No module named 'ompl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/share/ompl/demos/OptimalPlanning.py", line 49, in <module>
    from ompl import util as ou
ModuleNotFoundError: No module named 'ompl

DXXS, 2025-07-27:
FWIW, In the root directory of 'ompl' is a subdirectory 'py-bindings' which appears to have separate cmake capabilities; however if I issue a "cmake ." in that directory, it issues an error "could not find requested file: PythonBindingsUtils".

saahu27, 2025-07-28:
> brew install castxml for  Could NOT find castxml (missing: CASTXML)

DXXS, 2025-07-28:
? I'm not clear about your above message!? the claimed 'OPTIONAL' castxml appears to be ~huge :: is taking a ~LONG time to install (over 15 minutes now)...  however, it is also claiming not to be able to find 'Python', which I obviously executed in the second message above

Above build ~results in:

...
[100%] Linking CXX executable demo_MultiLevelPlanningHyperCubeBenchmark
[100%] Built target demo_MultiLevelPlanningHyperCubeBenchmark
(base) ds@Ds-MacBook-Air Release % sudo make install
Password:
Consolidate compiler generated dependencies of target ompl
[ 67%] Built target ompl
...
Install the project...
-- Install configuration: "Release"
...
-- Up-to-date: /usr/local/share/man/man1/plannerarena.1
(base) ds@Ds-MacBook-Air Release % python -m ompl
/Users/ds/.pyenv/versions/3.10.4/bin/python: No module named ompl
(base) ds@Ds-MacBook-Air ~ % which python
/Users/ds/.pyenv/shims/python
(base) ds@Ds-MacBook-Air ~ % python --version
Python 3.10.4

saahu27, 2025-07-28:
>Ah, you're using pyenv! This is a common issue when building OMPL with pyenv-managed Python

```
# Get the real Python path (not the shim)
PYTHON_EXEC=$(pyenv which python3)
echo $PYTHON_EXEC  # Should show something like /Users/ds/.pyenv/versions/3.10.4/bin/python3
PYTHON_ROOT=$(pyenv prefix)

# Configure OMPL with explicit paths
cmake  -DPYTHON_EXEC=$PYTHON_EXEC \
      -DPython3_EXECUTABLE=$PYTHON_EXEC \
      -DPython3_ROOT_DIR=$PYTHON_ROOT \
      ../..
```

saahu27, 2025-07-28:
> > If this is a 'common issue', then I'm wondering why haven't you fixed it yet? (IMO Macintosh users, at least, aren't generally likely to appreciate this level of ~offloading). My 5th message above clearly showed the python path is via pyenv... I'm still waiting for 'castxml' to complete installation, ~1.5 hours later...
> 
> This is not an issue as per ompl but an enhancement because its user specific feature. We are acively working towards rewriting the python bindings and will be released soon.
> 
> As per castxml, it is a very small package, it shouldn't take that long. can you please google a bit to find a better way to install castxml in mac.

DXXS, 2025-07-28:
Ok, it only took ~4 hours to install castxml... make with 'update_bindings' appeared to complete, and python appears to think that _something_ ompl has been installed; however, python3 now seems to have errors in executing the demos:

(base) ds@Ds-MacBook-Air ~ % python3 /usr/local/share/ompl/demos/DubinsAirplane.py
Traceback (most recent call last):
  File "/usr/local/share/ompl/demos/DubinsAirplane.py", line 41, in <module>
    from ompl import util as ou
  File "/Users/ds/.pyenv/versions/3.10.4/lib/python3.10/site-packages/ompl/util/__init__.py", line 4, in <module>
    from ompl.util._util import *
ModuleNotFoundError: No module named 'ompl.util._util'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/share/ompl/demos/DubinsAirplane.py", line 49, in <module>
    from ompl import util as ou
  File "/Users/ds/.pyenv/versions/3.10.4/lib/python3.10/site-packages/ompl/util/__init__.py", line 4, in <module>
    from ompl.util._util import *
ModuleNotFoundError: No module named 'ompl.util._util'

DXXS, 2025-07-28:
Your build log indicates that it's having difficulties finding stdio.h, wchar.h, and unistd.h;

However, in querying clang, the appropriate include paths are in the default search list.

In checking your CMake setup, it looks like you deliberately mess with stdio (esp. in the case of clang)

I think you should add Mac OSX to your QA/testing of compilation : from what I've read, docker now supports OSX ~images.

DXXS, 2025-07-28:
WTF?

Although I haven't been using docker on my Mac due to its own conflicts, NONE of those images appear to be appropriate for testing OSX compilation as they all are reportedly Linux images.

================================================================================

Primary errors in your log output appear to include.
...
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/stdlib.h:93:15: fatal error: 'stdlib.h' file not found
   93 | #include_next <stdlib.h>
      |               ^~~~~~~~~~
1 error generated.
...
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/wchar.h:119:15: fatal error: 'wchar.h' file not found
  119 | #include_next <wchar.h>
      |               ^~~~~~~~~
1 error generated.
...
In file included from /Users/ds/Development/miniforge3/include/boost/config/platform/macos.hpp:28:
/Users/ds/Development/miniforge3/include/boost/config/detail/posix_features.hpp:18:15: fatal error: 'unistd.h' file not found
   18 | #     include <unistd.h>
      |               ^~~~~~~~~~
1 error generated.
...
create_xml_file
    raise RuntimeError(
RuntimeError: Error occurred while running CASTXML xml file does not exist
-- Build type: Release

Clang reports that the appropriate search paths should be available:
================================================================================
#include <...> search starts here:
 /usr/local/include
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/13.0.0/include
 /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include
 /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks (framework directory)
End of search list.
================================================================================

Examining those search paths shows that the files are there:
================================================================================
(base) ds@Ds-MacBook-Air Release % find /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include -name stdlib.h
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/stdlib.h
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdlib.h
(base) ds@Ds-MacBook-Air Release % find /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include -name unistd.h
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/unistd.h
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/unistd.h
(base) ds@Ds-MacBook-Air Release % find /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include -name wchar.h
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__support/solaris/wchar.h
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/wchar.h
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/wchar.h
================================================================================

I think you should fix, and/or provide better errors.

saahu27, 2025-07-28:
This is a classic macOS header search path issue that occurs when #include_next directives can't find the next occurrence of system headers. The problem is that while the headers exist in the SDK, the compiler isn't properly configured to find them when processing #include_next.

1. Set the SDK path explicitly
First, check which SDK you're using:
```
xcrun --show-sdk-path
```
Then export the SDKROOT environment variable:
```
export SDKROOT=$(xcrun --show-sdk-path)
```

2. Ensure Xcode Command Line Tools are properly installed
Sometimes the Command Line Tools need to be reinstalled:
```
sudo rm -rf /Library/Developer/CommandLineTools
xcode-select --install
```

3. For CASTXML specifically
Since this appears to be a CASTXML build issue, you may need to pass additional flags:
```
# Set these environment variables before running CASTXML
export CPATH="$SDKROOT/usr/include"
export C_INCLUDE_PATH="$SDKROOT/usr/include"
export CPLUS_INCLUDE_PATH="$SDKROOT/usr/include/c++/v1:$SDKROOT/usr/include"
```

4. Check for conflicting installations
If you're using conda/miniforge (I see /Users/ds/Development/miniforge3 in your paths), there might be conflicts:
```
# Temporarily deactivate conda
conda deactivate

# Or set the compiler to use system headers first
export CFLAGS="-isysroot $(xcrun --show-sdk-path)"
export CXXFLAGS="-isysroot $(xcrun --show-sdk-path)"
```

DXXS, 2025-07-29:
• Yes, I had checked 'xcrun --show-sdk-path'; however, this is reportedly a ~naive way of generating system search paths, instead of building off that, it's reportedly better to get the whole explicated list of them, which have already been set, via:

% echo | clang -E -Wp,-v -

[ However, your ~brute-force explicit setting of SDKROOT + compiler paths does appear to have eliminated include errors! ]

• I had already verified the command line tools multivariantly, incl. in xcode.

• I was wondering why 'miniforge3' was showing up in the file paths, which is why I included it in the error log above, however, I'm not using conda at the moment, and your suggestion of 'conda deactivate' did not affect their inclusion in log results.  (Re)installing boost via brew didn't affect where its files are being included from, either (!?).  It looks like cmake is making some assumptions, and your cmake files aren't setting up SDKROOT at all!?

(base) ds@Ds-MacBook-Air ompl % cmake -G Xcode -S . -B build
...
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
-- Found Boost: /Users/ds/Development/miniforge3/lib/cmake/Boost-1.77.0/BoostConfig.cmake (found suitable version "1.77.0", minimum required is "1.68") found components: serialization filesystem system program_options 
...

DXXS, 2025-07-29:
I tried a couple of demos which seem to work;

However, in now attempting to compile genesis-world with your ~library, am receiving an error:

ERROR: Could not find a version that satisfies the requirement ompl>=1.7.0; platform_system != "Windows" (from genesis-world) (from versions: none)
ERROR: No matching distribution found for ompl>=1.7.0; platform_system != "Windows"

Not sure how their build systems may be checking; in checking myself via jupyter, it appears that you haven't implemented '__version__'!?

================================================================================

import ompl
print(ompl.__version__)
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[1], line 2
      1 import ompl
----> 2 print(ompl.__version__)

AttributeError: module 'ompl' has no attribute '__version__'

saahu27, 2025-07-29:
@DXXS If you have cloned the ompl from github, the release version is set to next release number. that is 1.8.0. If genesis requires 1.7.0 , please checkout the repo to 1.7.0 branch. else download the source code from [here](https://github.com/ompl/ompl/releases/tag/1.7.0)