ompl/ompl #1315: Build failures since merge of VAMP

rhaschke, 2025-08-28:
I'm facing a cmake failure since a1c467e5f3a5cfe57c4da16cf77c85e4be01a0e9 was merged:
```
cmake -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON -DFETCHCONTENT_FULLY_DISCONNECTED=ON -DCMAKE_INSTALL_RUNSTATEDIR=/run "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu -DBUILD_TESTING:BOOL=OFF -DCMAKE_INSTALL_PREFIX=/opt/ros/one -DCMAKE_PREFIX_PATH=/opt/ros/one -DBUILD_TESTING=OFF ../ompl

...
CMake Error at external/vamp/cmake/Dependencies.cmake:38 (add_library):
  Cannot find source file:

    /tmp/ompl.build/_deps/simdxorshift-src/src/simdxorshift128plus.c
Call Stack (most recent call first):
  external/vamp/CMakeLists.txt:53 (include)

CMake Error at external/vamp/cmake/Dependencies.cmake:38 (add_library):
  No SOURCES given to target: simdxorshift
Call Stack (most recent call first):
  external/vamp/CMakeLists.txt:53 (include)
```
Note that GHA workflows fail since then as well: https://github.com/ompl/ompl/actions?query=branch%3Amain, however for unrelated reasons.
The culprit for me seems to be the cmake flag `-DFETCHCONTENT_FULLY_DISCONNECTED=ON`, which seems to be injected by `bloom-generate`. If I remove that flag, I works locally.

mamoll, 2025-08-28:
Thanks for the report, @rhaschke! There have been a number of surprises that didn't show up in internal testing.

@saahu27, can you follow up?

saahu27, 2025-08-29:
> Thanks for the report, @rhaschke! There have been a number of surprises that didn't show up in internal testing.
> 
> @saahu27, can you follow up?

Yes sir, I am working on the issue.

rhaschke, 2025-09-17:
> The culprit for me seems to be the cmake flag `-DFETCHCONTENT_FULLY_DISCONNECTED=ON`, which seems to be injected by `bloom-generate`.

Turned out that this cmake flag is added by debhelper since recently:
https://sources.debian.org/src/debhelper/13.26/debian/changelog#L710

However, there remain build issues on ARM platforms:
```
In file included from /<<BUILDDIR>>package/external/vamp/src/impl/vamp/vector.hh:9:
/<<BUILDDIR>>package/external/vamp/src/impl/vamp/vector/neon.hh: In static member function ‘static constexpr vamp::SIMDVector<__Int32x4_t>::VectorT vamp::SIMDVector<__Int32x4_t>::cmp_not_equal(VectorT, VectorT)’:
/<<BUILDDIR>>package/external/vamp/src/impl/vamp/vector/neon.hh:70:51: note: use ‘-flax-vector-conversions’ to permit conversions between vectors with differing element types or numbers of subparts
   70 |             return vreinterpretq_s32_u32(vmvnq_s32(vceqq_s32(l, r)));
      |                                          ~~~~~~~~~^~~~~~~~~~~~~~~~~
/<<BUILDDIR>>package/external/vamp/src/impl/vamp/vector/neon.hh:70:61: error: cannot convert ‘uint32x4_t’ to ‘int32x4_t’
   70 |             return vreinterpretq_s32_u32(vmvnq_s32(vceqq_s32(l, r)));
      |                                                    ~~~~~~~~~^~~~~~
      |                                                             |
      |                                                             uint32x4_t

```

saahu27, 2025-10-01:
@rhaschke The arm issue is resolved in main now. Can we close this issue?

rhaschke, 2025-10-01:
Thanks for pointing this out. Yes, my nightly build succeeds now.

closed 2025-10-01 by rhaschke